prusa3d/PrusaSlicer #5572: Volumetric flow rate is very slow

djr7C4, 2020-12-24:
### Version
2.3.0 rc2

### Operating system type + version
Arch Linux, kernel 5.9.6

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
Switching to show the volumetric flow rate is very slow for long prints.  This was not a problem in the 2.2.0 series.  The program is also slow to respond when adjusting the top layer shown after switching to volumetric flow rate.

#### Project File (.3MF) where problem occurs
[vf-slow.zip](https://github.com/prusa3d/PrusaSlicer/files/5737981/vf-slow.zip)

bubnikv, 2021-01-04:
Isn't it due to Ironing? If you disable ironing, do you have the same issue?

djr7C4, 2021-01-06:
Ironing is already disabled in the 3mf file I tested it on.  See the zip archive above.

bubnikv, 2021-01-06:
I see it now. Switching the G-code preview to volumetric speed visualization makes the vertical slider selecting the range of layers move very slowly.

bubnikv, 2021-01-06:
[image]

The bottleneck is 
Slic3r::GUI::GCodeViewer::refresh_render_paths()
which is slow and it is also being called twice on vertical slider update, while once would suffice.

bubnikv, 2021-01-06:
Significantly improved with https://github.com/prusa3d/PrusaSlicer/commit/daa0bbdb0cfad984565bbd4a17940a65ef28650e
It is not smooth, but usable.
Thanks for the report.

closed 2021-01-06 by bubnikv