prusa3d/PrusaSlicer #5573: PrusaSlicer won't launch

schneidp20, 2020-12-24:
### Version
2.3.0 rc2,   2.2.9.1

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 Version 2004

### 3D printer brand / version + firmware version (if known)
Prusa MK3, but not really applicable in this case

### Behavior
Slicer will only launch when just running on my laptop.  When 2 external monitors are used, slicer won't launch.   External monitors are 1920x1080, laptop monitor is 3840x2400.  I dropped the laptop resolution down to match the monitors, but slicer still doesn't launch.  Also closing the laptop forces just the low res monitors to be used and slicer still won't launch.  The work around is to launch slicer before connecting the monitors.  When this is done, slicer can maximize, but when not maximized, slicer won't resize small enough to fit on the screen.  It's like slicer is using the laptop display resolution to determine its size even when the laptop monitor is off.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Kachidoki2807, 2020-12-24:
Hello,
What do you mean by "won't launch"? Do you see the splashcreen? Do you have a crash? Do you have nothing happenning?
I am currently using PS 2.3.0-rc2 with a dual monitor (2560x1440 and 1680x1050) under Win10 2004 without issue.

schneidp20, 2020-12-24:
I see the hourglass for a moment and then nothing (no splash screen).  I
checked the task manager thinking it was just off-screen somewhere, but
no.  It must have crashed. Is there a log file somewhere?  I have 40 years
of software engineering under my belt so if there is something I can look
for or monitor I'd be glad to help.  I'm not sure I want to sign up to fix
it myself though.

On a side note, if I use my work laptop with the same docking station,
slicer runs fine.  In that case all 3 monitors have same default resolution
(1680x1050).  My guess it has something to do with the native resolution of
my personal laptop.  Like I said in the incident, I dropped the resolution
of my personal laptop to match the monitors, rebooted with same results.

THanks,
David

schneidp20, 2020-12-24:
More system info.
[image: image.png]

horviskoadrianko, 2020-12-30:
I have the same issue. It just crashes when opened on seccond monitor.

obgr, 2021-01-02:
I have a similar issue which seems to be linked to multiple displays.

**PrusaSlicer version**
PrusaSlicer-2.3.0-rc2+win64-202012221641

**Operating system type + version**
Windows 10 (build 19042), 64-bit edition (freshly installed).

**Other system information + Hardware**
OpenGL installation
GL version: 4.6.0 NVIDIA 460.89
Vendor: NVIDIA Corporation
Renderer: GeForce RTX 3070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
CPU: Ryzen 3700x
Monitors: 2 monitors (both at 1440p)

**Problem**
After downloading Prusaslicer 2.3 RC-2 and attempting to run it, process hangs and responds as "Not responding" in task manager.
Splash screen shows up, but application hangs while loading.

[image]

**Work Around**
1. Disconnect secondary display.
2. Start prusaslicer.

Prusaslicer does not start on the secondary display like @horviskoadrianko mentioned.

bubnikv, 2021-01-04:
Maybe the dreaded Nahimic issue?

https://github.com/prusa3d/PrusaSlicer/issues/2522
https://github.com/prusa3d/PrusaSlicer/issues/5091
https://github.com/prusa3d/PrusaSlicer/issues/4793
https://github.com/prusa3d/PrusaSlicer/issues/3689

bubnikv, 2021-01-04:
https://github.com/prusa3d/PrusaSlicer/issues/2142

If this is the Nahimic driver issue, then please bug https://www.nahimic.com/

schneidp20, 2021-01-04:
I don't know.   I can't find Nahimic anywhere.  It's not something that
shows up as an application/feature and I don't see it in the running
services.  I also searched for it on my SSD.

[image: image.png]

bubnikv, 2021-01-04:
Is your motherboard or laptop Asus made?

schneidp20, 2021-01-04:
No, unless Dell farms out motherboard work to Asus.  It's a Dell XPS laptop.

bubnikv, 2021-01-04:
Thanks. Unfortunately these issues are difficult to solve without having access to such a device.

schneidp20, 2021-01-04:
My other business laptop is a Dell Precision 5540.   That one, like I said,
has the same resolution as my external monitors and Slicr works just fine
with the same docking station.

bubnikv, 2021-01-04:
> I'm a S/W guy so I totally understand. I guess if it bugs me enough, I can
> build and debug the problem myself. I have a work-around though ... start
> Slicr before docking my laptop. Thanks

> I'm a S/W guy so I totally understand. I guess if it bugs me enough, I can
build and debug the problem myself.

That would help immensely.

schneidp20, 2021-01-04:
I read your "How to Build" document.  Should I be using VS2019?  I'm It
sounds like you are in a state of flux going from VS2013 to VS2019. It
would be in your best interest to keep the how-to current.    Thanks

obgr, 2021-01-04:
@bubnikv 
I have never heard about Nahimic but yes, **i found** a **service running** on my machine. It must be included in the realtek drivers from my motherboard manufacturer (Asus).
After stopping the service, prusaslicer starts as expected on my secondary monitor.

Since i cannot find any newer drivers, i'll try to get rid of nahimic.
**But before i do**, would you like me to debug something for you? I have compiled a version of prusaslicer 2.3.0 rc-2 that i prepared in order to troubleshoot this. 
Please note that i am not really a software dev, so i might need some instructions to get you what you need.
Br

schneidp20, 2021-01-04:
Olle,
We might be chasing different issues.  Mine is on a Dell w/o any Nahimic.
I'm a S/W developer, but this is new territory for me, If you have any tips
on setting the environment I'm all ears!  :-)

obgr, 2021-01-04:
@schneidp20 
I did not think i was running any Nahimic software at first either, so i suspected the issues were related. Later on i found a service named "NahimicService" which was listed in services.msc.
I would take an extra look just to be sure, Nahimic lists Dell as a vendor on their website.

**Regarding Dev environment.**
I just followed the [How to build - Windows.md noob-guide-step-by-step](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md#noob-guide-step-by-step) guide.
I had two issues. 

1. When building the dependencies, git needs to be in the environment variable "path", i had to manually add it. Also make sure to build the dependencies via a terminal opened via Start Menu and Click on "Visual Studio 2019" folder, then select the ->"x64 Native Tools Command Prompt" like the instructions say, otherwise the environment vars might load correctly.
2. First time i built the actual prusaslicer solution, something failed and prusaslicer-console reported that prusaslicer.dll was not loaded when started. I am not sure what went wrong, i just performed a cleanup and rebuilt the solution and prusaslicer started as expected in debug mode.

Hope it helps.

bubnikv, 2021-01-05:
@obgr

> I have never heard about Nahimic but yes, i found a service running on my machine. It must be included in the realtek drivers from my motherboard manufacturer (Asus).
> After stopping the service, prusaslicer starts as expected on my secondary monitor.

Cool.

BTW
https://rog.asus.com/forum/showthread.php?107802-Nahimic-Service-causing-programs-to-startup-slow
and possibly many others

This may be a quick test, that anybody could try. Just run the following from the command line:

```
taskkill /f /im "NahimicService.exe"
taskkill /f /im "NahimicSvc32.exe"
taskkill /f /im "NahimicSvc64.exe"
```

> Since i cannot find any newer drivers, i'll try to get rid of nahimic.

You may also try to bug Asus and/or Nahimic. I am really interested in their response.

> But before i do, would you like me to debug something for you? I have compiled a version of prusaslicer 2.3.0 rc-2 that i prepared in order to troubleshoot this.

Frankly I don't know. Is it crashing? Or hanging? I remember from analyzing some crash logs that it was crashing inside either the Nahimic or OpenGL driver.

It would be best if we can get hands on one of these computers and debug it ourselves.

bubnikv, 2021-01-05:
process explorer 
https://docs.microsoft.com/en-us/sysinternals/downloads/process-explorer
shows list of loaded DLLs into a process. The issue with the Nahimic service is that it injects a DLL into PrusaSlicer process, which somehow interferes with OpenGL. Maybe we can somehow block the Nahimic service to inject the Nahmic DLL. It is likely technically possible, but it would require some experiments and access to the Nahimic plagued machine would be needed.

We can likely detect the Nahimic DLL loaded into our PrusaSlicer process and report it, possibly with a notification or in System info. That may give user a hint, but it does not really help to resolve the issue.

bubnikv, 2021-01-05:
It is not just Asus, but also MSI that preinstalls the Nahimic drivers.

bubnikv, 2021-01-05:
Most likely the Nahimic service injects a DLL using 1, 5 or 6 method described in 
https://www.elastic.co/blog/ten-process-injection-techniques-technical-survey-common-and-trending-process
The method 1 is detailed in 
https://dzone.com/articles/windows-api-hooking-and-dll-injection

It would help us, if somebody documents the names of the DLLs injected into PrusaSlicer by Nahimic service and possibly documents the way the DLL is injected. A "fix" may consist of creating phony Nahimic DLLs and loading them into PrusaSlicer before Nahimic, however I am not sure how reliable this would be. 

https://github.com/microsoft/Detours
may be used to hook the LoadLibraryA/LoadLibraryW/LoadLibraryExA/LoadLibraryExW Win32 API functions to refuse to load the Nahimic DLL. That may help if the Nahimic service is using the method 1 to inject the DLL. That may be worth trying.

obgr, 2021-01-05:
@bubnikv 
If you want to move this to another Issue, please tell me and i can open a new one.

> Frankly I don't know. Is it crashing? Or hanging? I remember from analyzing some crash logs that it was crashing inside either the Nahimic or OpenGL driver.

I do not really know since i have yet to see any logs that tells me anything, event viewer is not helping much. but it looks like it is hanging. prusa-slicer-console is showing nothing, the process is listed in taskmanager but in a "not responding" state. I included some debug output from when i attempted to start prusaslicer and later on killed the task.
<details>
  <summary>Debug output from visual studio</summary>

```
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Users\Olle\Downloads\PrusaSlicer-version_2.3.0-rc2\build\src\Debug\prusa-slicer-console.exe'. Symbols loaded.
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140d.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140d.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140_1d.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbased.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. 
The thread 0x4c58 has exited with code 0 (0x0).
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\uxtheme.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\msctf.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\A-Volute.SonicRadar3DevProps2.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\ProductInfo.dll'. 
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
'prusa-slicer-console.exe' (Win32): Loaded 'C:\ProgramData\A-Volute\A-Volute.SonicStudio3\Modules\ScheduledModules\x64\A-Volute.SonicStudio3DevProps2.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Program Files (x86)\RivaTuner Statistics Server\RTSSHooks64.dll'. Module was built without symbols.
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\winmm.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\opengl32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\glu32.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\clbcatq.dll'. 
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\AppXDeploymentClient.dll'. 
'prusa-slicer-console.exe' (Win32): Unloaded 'C:\Windows\System32\AppXDeploymentClient.dll'
'prusa-slicer-console.exe' (Win32): Loaded 'C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_a51067c0ac557884\nvoglv64.dll'. 
[... cut ...]
| A-Volute.SonicStudio3DevProps2.dll | | A-Volute | C:\ProgramData\A-Volute\A-Volute.SonicStudio3\Modules\ScheduledModules\A-Volute.SonicStudio3DevProps2.dll | |
| NahimicOSD.dll | NahimicOSD | Nahimic | C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\NahimicOSD.dll | |
| ProductInfo.dll | | | C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\ProductInfo.dll | |
| ProductInfo.dll | | | C:\ProgramData\A-Volute\A-Volute.SonicStudio3\Modules\ScheduledModules\ProductInfo.dll | Not loaded in prusa-slicer-console, but loaded in another 3d application |

Is there anything that you like me to look further in?
BR

bubnikv, 2021-01-05:
@obgr Would you please check whether the DLL is referenced in registry? See the section 6 of 
https://www.elastic.co/blog/ten-process-injection-techniques-technical-survey-common-and-trending-process

obgr, 2021-01-05:
@bubnikv 
No, i cannot find any of the listed dlls referenced in the registry.
BR

bubnikv, 2021-01-05:
@obgr maybe we will put together an experimental build patching the LoadLibrary API to filter out the Nahimic DLL, but it will not be soon.

obgr, 2021-01-05:
@bubnikv i'd be happy to help.
Mention me when the time comes and i'll do my part.
as a work around, i'll disable the service.

kocikdav, 2021-01-20:
@obgr Hello I've prepared a build that is printing to console all dlls loaded to PrusaSlicer during runtime. Would you please run it on your computer and give us the result?
https://www.dropbox.com/s/ljsnm1t483e9zmo/PrusaSlicer-2.3.0%2B40-dk-injection-win64-g5a81c8e86-202101200857.zip?dl=0

schneidp20, 2021-01-20:
Sure.   I'm not home right now, but I'll give it a try this evening (I'm
GMT-6)

Thanks!

obgr, 2021-01-20:
@kocikdav 

here is the output i get from prusaslicer-console.
<details>
  <summary>Nahimic service stopped</summary>
  
```
LoadLibraryExA: secur32.dll
LoadLibraryW: user32.dll
LoadLibraryExW: User32.dll
LoadLibraryExW: Shcore.dll
LoadLibraryExW: User32.dll
LoadLibraryW: ntdll.dll
LoadLibraryA: hid.dll
```
</details>

<details>
  <summary>Nahimic service running</summary>
  
```
LoadLibraryExA: secur32.dll
LoadLibraryW: user32.dll
LoadLibraryExW: User32.dll
LoadLibraryExW: Shcore.dll
LoadLibraryExW: User32.dll
LoadLibraryW: ntdll.dll
LoadLibraryA: hid.dll
LoadLibraryW: d3d9.dll
LoadLibraryW: C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_a51067c0ac557884\nvldumdx.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExW: ext-ms-win-core-resourcepolicy-l1-1-0.dll
LoadLibraryW: C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_a51067c0ac557884\nvldumdx.dll
LoadLibraryA: gdi32.dll
```
</details>

<details>
  <summary>Nahimic service running started on secondary screen</summary>
  
```
LoadLibraryExA: secur32.dll
LoadLibraryW: user32.dll
LoadLibraryExW: User32.dll
LoadLibraryExW: Shcore.dll
LoadLibraryExW: User32.dll
LoadLibraryW: ntdll.dll
LoadLibraryA: hid.dll
LoadLibraryW: d3d9.dll
LoadLibraryW: C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_a51067c0ac557884\nvldumdx.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExW: ext-ms-win-core-resourcepolicy-l1-1-0.dll
LoadLibraryW: C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_a51067c0ac557884\nvldumdx.dll
LoadLibraryA: gdi32.dll
```
</details>

Give me a shout if you need anything else!
BR

bubnikv, 2021-01-20:
@obgr 
Thanks. Unfortunately as of now it seems the workaround we had in mind did not help.

Are you able to check the system registry for us? See the 6th way of injecting a DLL into a process. Do you find the nahimic dll mentioned in that section in your registry?
https://www.elastic.co/blog/ten-process-injection-techniques-technical-survey-common-and-trending-process

obgr, 2021-01-20:
> @obgr
> Thanks. Unfortunately as of now it seems the workaround we had in mind did not help.
>
> Are you able to check the system registry for us? See the 6th way of injecting a DLL into a process. Do [...]

Hi
I'm afraid i do not find anything of interest on the specified places in the registry.
If you wish, i could share my desktop over slack, teams, discord or something if you want to take a closer look in real time.
Br

schneidp20, 2021-01-21:
I'm not sure what to do.  I tried running prusaslicer from the command
prompt, but got nothing.  I'm not sure where to look for the console output.

schneidp20, 2021-01-21:
Ok, I get it.  Below is what I get on my laptop as a stand-alone w/o the
docking station monitors.   With the monitors attached, the
prusaslicer console briefly opens, shows nothing at all, and then closes.

LoadLibraryExA: secur32.dll
LoadLibraryW: user32.dll
LoadLibraryExW: User32.dll
LoadLibraryExW: Shcore.dll
LoadLibraryExW: User32.dll
LoadLibraryW: ntdll.dll
LoadLibraryA: hid.dll
LoadLibraryExW: dxgi.dll
LoadLibraryExW: d3d11.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExW: msasn1.dll
LoadLibraryExW: cryptnet.dll
LoadLibraryExW: cryptbase.dll
LoadLibraryExW: crypt32.dll
LoadLibraryExW: wintrust.dll
LoadLibraryExW:
C:\Windows\System32\DriverStore\FileRepository\nvdm.inf_amd64_090da5d9229c44ea\NvCamera\NvCameraWhitelisting64.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: api-ms-win-appmodel-runtime-l1-1-2
LoadLibraryExW: api-ms-win-appmodel-runtime-l1-1-2
LoadLibraryExW: api-ms-win-appmodel-runtime-l1-1-2
LoadLibraryExW: api-ms-win-core-sysinfo-l1-2-1
LoadLibraryExW: api-ms-win-appmodel-runtime-l1-1-2
LoadLibraryExW: api-ms-win-appmodel-runtime-l1-1-2
LoadLibraryExW: api-ms-win-appmodel-runtime-l1-1-2
LoadLibraryExW: user32.dll
LoadLibraryExW: nvapi64.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExW: msasn1.dll
LoadLibraryExW: cryptnet.dll
LoadLibraryExW: cryptbase.dll
LoadLibraryExW: gdi32.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryW: Combase.dll
LoadLibraryExA:
C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_304b99360206b832\igdgmm64.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExA:
C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_304b99360206b832\igc64.dll
LoadLibraryExW: api-ms-win-core-synch-l1-2-0
LoadLibraryExW: api-ms-win-core-fibers-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-l1-2-1
LoadLibraryExW: kernel32
LoadLibraryExW: api-ms-win-core-string-l1-1-0
LoadLibraryExW: api-ms-win-core-datetime-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-obsolete-l1-2-0
LoadLibraryExW: api-ms-win-core-string-l1-1-0
LoadLibraryExW: api-ms-win-core-datetime-l1-1-1
LoadLibraryExW: api-ms-win-core-localization-obsolete-l1-2-0
LoadLibraryW: winmm.dll

schneidp20, 2021-01-21:
You guys might be off the hook for this problem.   I'm loading up my new
laptop only to find that Fusion 360 has the same symptoms.  I can only
start the app w/o being docked.  I can then redock with Fusion 360 running
and then work with my docking station monitors.  I'll create an incident
with Fusion 360 and let you know what they say about the problem.

kocikdav, 2021-01-22:
Thank you @obgr @schneidp20 for testing. Looks like Nahimic dll is not loaded dynamically during runtime. I have prepared another build that enumerates all dlls at several places during startup. Would you please run it for us? It should print total nuber of dlls and if NahimicOSD.dll is present for each control point. 
https://www.dropbox.com/s/7tfczz79qpqdec6/PrusaSlicer-2.3.0%2B42-dk-injection-win64-g6cac9d739-202101221549.zip?dl=0

obgr, 2021-01-22:
@kocikdav
Here is the output of the console.

<details>
  <summary>Nahimic service stopped</summary>
  
```
control point 1
Total Dlls: 68

control point 2
Total Dlls: 71

control point 3
Total Dlls: 71

control point 4
Total Dlls: 75

control point 5
Total Dlls: 75

control point 6
Total Dlls: 82
```
</details>

<details>
  <summary>Nahimic service on, prusalicer started on primary screen</summary>

```
control point 1
Total Dlls: 71
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 2
Total Dlls: 74
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 3
Total Dlls: 74
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 4
Total Dlls: 78
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 5
Total Dlls: 78
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 6
Total Dlls: 85
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll
```
</details>

<details>
  <summary>Nahimic service on, prusalicer started on secondary screen</summary>

```
control point 1
Total Dlls: 71
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 2
Total Dlls: 74
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 3
Total Dlls: 74
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 4
Total Dlls: 78
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 5
Total Dlls: 78
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

control point 6
Total Dlls: 85
Contains library: C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll
```

</details>

schneidp20, 2021-01-23:
Same behavior as before.  When docked, I get nothing in the console before
it crashes.  As a stand-alone laptop, I get the following.   Thanks!
control point 1
Total Dlls: 65

control point 2
Total Dlls: 68

control point 3
Total Dlls: 68

control point 4
Total Dlls: 72

control point 5
Total Dlls: 72

control point 6
Total Dlls: 80

kocikdav, 2021-02-01:
Thank you both for testing. Here's another build. It should display proper warning dialog about presence of NahimicOSD dll. If there is no error in the code, It should surely display for @obgr . 
https://www.dropbox.com/s/on0dwea75gfx4cc/PrusaSlicer-2.3.0%2B127-win64-gf939cac84-202102011345.zip?dl=0
We know that this is just a warning rather than solution, but currently we have no way to stop loading this library as it comes linked statically when we start our process.

@schneidp20 as you mentioned before, your problem might be somewhere else. Have you succeeded with building PrusaSlicer on your PC? There is a Step by step guide (should be up to date now) you can follow. It would help us greatly.

obgr, 2021-02-01:
@kocikdav Warning message is displayed upon start of prusaslicer.
[image]

Br Olle

schneidp20, 2021-02-09:
Sorry for the delay.   Sorry, but same symptom as before.  There is no
output at all.  Both the stand-alone app and the console app die w/o
displaying anything.

Here is another bit of info. As you know, my work-around is to launch
Slicer before docking my laptop.  The interesting part is I accidentally
launched a second instance of the app after docking, and that launch works
fine.

No I haven't tried to compile and debug it myself. I also haven't
contacted Fusion 360 about a similar problem..

schneidp20, 2021-02-10:
FYI: I got a reply from Fusion 360 customer support.   Sad, but their reply
was to start the application with one monitor ... the work-around I already
was doing for both Slicer and Fusion 360. :-(

jozsefhabit, 2021-03-01:
Hi, I have also a Dell XPS noteboook, and have the same symptoms that @schneidp20 mentioned. 
I gonna compile the code and let you know the results of the debug process

bubnikv, 2021-03-01:
> FYI: I got a reply from Fusion 360 customer support. Sad, but their reply
was to start the application with one monitor ...

They are my heroes :-)

jozsefhabit, 2021-03-03:
I have been able to build PrusaSlicer on my laptop. I got this debug output when running it with the two monitors attached and with different resolutions:

`'prusa-slicer.exe' (Win32): Loaded 'C:\tools\src\PrusaSlicer\build\src\Debug\prusa-slicer.exe'. Symbols loaded.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140_1d.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140d.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140d.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbased.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\uxtheme.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msctf.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\opengl32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\glu32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\clbcatq.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\AppXDeploymentClient.dll'. 
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\AppXDeploymentClient.dll'
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_5f148aca851713ad\ig9icd64.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\dwmapi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\dxgi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ResourcePolicyClient.dll'. 
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\ResourcePolicyClient.dll'
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\wldp.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\DriverStore\FileRepository\nvdmi.inf_amd64_d4f5fb7711ba6c16\nvdlistx.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. 
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\version.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\DriverStore\FileRepository\nvdmi.inf_amd64_d4f5fb7711ba6c16\nvdlistx.dll'
Exception thrown at 0x0000000000000000 in prusa-slicer.exe: 0xC0000005: Access violation executing location 0x0000000000000000.
`

Do you need any more info, or tests that I can help with?

jozsefhabit, 2021-03-03:
Here is the call stack:

0000000000000000()
ig9icd64.dll!00007ffa433cb8bf()
ig9icd64.dll!00007ffa433bd624()
ig9icd64.dll!00007ffa433a1904()
opengl32.dll!00007ffa91093c1b()
opengl32.dll!00007ffa910917a1()
opengl32.dll!00007ffa91093c7c()
opengl32.dll!00007ffa910afc90()
opengl32.dll!00007ffa910b0cb1()
opengl32.dll!00007ffa910b00b3()
gdi32full.dll!00007ffad6a72399()
prusa-slicer.exe!OpenGLVersionCheck::check(HWND__ * hWnd) Line 155
	at C:\tools\src\PrusaSlicer\src\PrusaSlicer_app_msvc.cpp(155)
prusa-slicer.exe!OpenGLVersionCheck::supports_opengl2_wndproc(HWND__ * hWnd, unsigned int message, unsigned __int64 wParam, __int64 lParam) Line 190
	at C:\tools\src\PrusaSlicer\src\PrusaSlicer_app_msvc.cpp(190)
user32.dll!00007ffad7a6e858()
user32.dll!00007ffad7a6e3dc()
user32.dll!00007ffad7a82bf7()
ntdll.dll!00007ffad8f10a44()
win32u.dll!00007ffad66b1ec4()
user32.dll!00007ffad7a67d8b()
user32.dll!00007ffad7a67958()
user32.dll!00007ffad7a677a2()
prusa-slicer.exe!OpenGLVersionCheck::load_opengl_dll() Line 59
	at C:\tools\src\PrusaSlicer\src\PrusaSlicer_app_msvc.cpp(59)
prusa-slicer.exe!wWinMain(HINSTANCE__ * __formal, HINSTANCE__ * __formal, wchar_t * __formal, int __formal) Line 245
	at C:\tools\src\PrusaSlicer\src\PrusaSlicer_app_msvc.cpp(245)
prusa-slicer.exe!invoke_main() Line 123
	at D:\a01\_work\9\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl(123)
prusa-slicer.exe!__scrt_common_main_seh() Line 288
	at D:\a01\_work\9\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl(288)
prusa-slicer.exe!__scrt_common_main() Line 331
	at D:\a01\_work\9\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl(331)
prusa-slicer.exe!wWinMainCRTStartup(void * __formal) Line 17
	at D:\a01\_work\9\s\src\vctools\crt\vcstartup\src\startup\exe_wwinmain.cpp(17)
kernel32.dll!00007ffad8cb7034()
ntdll.dll!00007ffad8ec2651()

jozsefhabit, 2021-03-03:
Ok, and to be sure, that the above crash output and stack trace is really related to two monitors with different resolutions, I have started a debug with two monitors in Duplicate mode (same resolution on both displays), instead of Extend mode. 
PrusaSplicer started smoothly, no errors, stack trace, and in the VisualStudio got this debug output:

'prusa-slicer.exe' (Win32): Loaded 'C:\tools\src\PrusaSlicer\build\src\Debug\prusa-slicer.exe'. Symbols loaded.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140d.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140_1d.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140d.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbased.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\uxtheme.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msctf.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\opengl32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\glu32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\wldp.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\DriverStore\FileRepository\nvdmi.inf_amd64_d4f5fb7711ba6c16\nvdlistx.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. 
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\version.dll'
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\DriverStore\FileRepository\nvdmi.inf_amd64_d4f5fb7711ba6c16\nvdlistx.dll'
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\clbcatq.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\AppXDeploymentClient.dll'. 
'prusa-slicer.exe' (Win32): Unloaded 'C:\Windows\System32\AppXDeploymentClient.dll'
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\DriverStore\FileRepository\nvdmi.inf_amd64_d4f5fb7711ba6c16\nvoglv64.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\setupapi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\bcrypt.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\wtsapi32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\msasn1.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\cryptnet.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\crypt32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\cryptbase.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\devobj.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\wintrust.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ntmarta.dll'. 
The thread 0x6b80 has exited with code 0 (0x0).
The thread 0x59ac has exited with code 0 (0x0).
The thread 0x54f8 has exited with code 0 (0x0).
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\DXCore.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\nvspcap64.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\profapi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\dwmapi.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\powrprof.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\umpdc.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\winsta.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\tools\src\PrusaSlicer\build\src\Debug\PrusaSlicer.dll'. Symbols loaded.
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\Wldap32.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\normaliz.dll'. 
[... 36 lines cut ...]
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\winmm.dll'. 
[... cut ...]
The thread 0x6df0 has exited with code 0 (0x0).
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\schannel.dll'. 
The thread 0x5ea8 has exited with code 0 (0x0).
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\mskeyprotect.dll'. 
'prusa-slicer.exe' (Win32): Loaded 'C:\Windows\System32\ncryptsslp.dll'. 
The thread 0x3abc has exited with code 0 (0x0).
The thread 0x6c0c has exited with code 0 (0x0).
The thread 0x5f6c has exited with code 0 (0x0).

@bubnikv let me know, if you need me to test anything else, please.

bubnikv, 2021-03-03:
@enricoturri1966 would you please investigate the call stacks?

the-other-shane, 2021-05-07:
My experience mirror's @jozsefhabit's, however I found that I can use external monitors in 'Extend' mode _provided my laptop is set as the 'main display' (set via Display Settings)._

So for those chasing a workaround - I find that having the laptop as my main display is less cumbersome than temporarily undocking or changing the external monitor projection mode in order to launch PrusaSlicer.

bubnikv, 2021-07-30:
We have one computer available in house, where PrusaSlicer hangs if a USB 3 based docking station is connected with specifically 2 monitors connected and it has to be specific monitors. With other 2 monitors PrusaSlicer starts just fine. We are investigating it further. It is supposed to work before the last big Windows 10 update, it broke after the Windows 10 update.

bubnikv, 2021-07-30:
We have a handful of ASUS based desktops around, but as for now we did not find any, for which the ASUS Sound Studio would be available.

obgr, 2021-07-30:
@bubnikv 
I am considering to switch motherboard. the Nahimic issue is one of the reasons but not the only one.
Would it be interesting for you to use that motherboard for troubleshooting if i replace it (only the motherboard itself)?

bubnikv, 2021-07-30:
@obgr Maybe. I would have to check with our IT department whether they could build a test setup around it.
What motherboard is it?

obgr, 2021-07-30:
@bubnikv 
Its a ROG STRIX B550-I GAMING. so you basically need a newer Ryzen cpu (3000 or 500 series) and a ddr4 ram stick.

bubnikv, 2021-09-25:
Some of the issues here may not be Nahimic related, but similar to https://github.com/prusa3d/PrusaSlicer/issues/2939

Damien-Ra, 2021-12-22:
Hello,

I've had the same problem for quite some time now, but was just made to look by the prompt on v2.4.

Stopping the Naimic service did solve the issue, but I also found this [link](https://chieftalk.chiefarchitect.com/topic/28384-multi-monitor-issue-fix-for-computers-with-nahimic-audio/) where they point to a BlackApps.dat file where you can add the executable that have the issue with this Nahimic services.

I tried it and it solve the problem for for me (with prusaslicer but also fusion 360 and openscad).
Also looking at the looong list of .exe already in the file I wonder how they can release such a driver …

Hope this helps !

MartinRusk, 2021-12-28:
The BlackApps.dat solution worked for me also. No more problems on my secondary monitor. This seems to be the proper solution. Now only the startup prompt is remaining. Perhaps a "do not show again" option on that dialogue would be nice to hide it in the future.

schneidp20, 2021-12-28:
Glad it worked for you, but there is no such file (BlackApps.dat) on my
system, plus I can't find  Nahimic  or any of the 'Sonic" apps listed in
the link.  It's not just Pusua though.  Fusion 360 has the same problem and
no response from their customer support.    Dell XPS 15 9500, Windows 10.
 I'm still stuck with the work-around of starting the apps un-docked.

krullshards, 2022-01-11:
Hi Guys, I'm having a similar issue to other folks here. I've never had an issue with Prusa Slicer but this holiday season I ended up picking up two monitors and extending my desktop. Now, my setup is a little bit different from some folks. I have these stacked vertically, not sure if that is what is causing my issue. When I start the slicer the splash screen displays and then disappears with no slicer window. I checked the windows event viewer in the application log and I saw that Prusa Slicer had crashed. I am running Prusa Slicer Version 2.4.0.

Here is the windows event log information:

Faulting application name: prusa-slicer.exe, version: 2.4.0.0, time stamp: 0x61c1feea
Faulting module name: PrusaSlicer.dll, version: 0.0.0.0, time stamp: 0x61d4237f
Exception code: 0xc0000005
Fault offset: 0x0000000001e53a1d
Faulting process id: 0xd78c
Faulting application start time: 0x01d8069670a5f5e3
Faulting application path: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
Faulting module path: C:\Program Files\Prusa3D\PrusaSlicer\PrusaSlicer.dll
Report Id: 610d98b8-3f81-42e9-82a1-9cc3d7356015
Faulting package full name: 
Faulting package-relative application ID: 

One interesting thing to note, if I delete "window_mainframe" property in the PrusaSlicer.ini file, I am able to launch the slicer until the next time after it writes its window position into the file again.

Hope that helps.

lukasmatena, 2022-01-14:
We tried to workaround this problem in upcoming 2.4.1 release by making the position restoring optional (and on by default). When the crash described here occurs, PrusaSlicer will suggest to turn the feature off on the next start. It is not exactly a fix, but it is better than nothing.

Because we cannot reproduce the issue, could you guys please test the following build and let us know if it helped?
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.4.1-alpha0+4-win64-g200cbd165-202201141630.zip
If the download requires login use login: prusaslicer password: slicer.

If the build crashes on the very first try, run it again. Thanks a lot.

MartinRusk, 2022-01-21:
Fusion 360 has the same problem on my System, same solution works. Location:
"C:\ProgramData\A-Volute\DellInc.AlienwareSoundCenter\Modules\ScheduledModules\Configurator\BlackApps.dat"

> Glad it worked for you, but there is no such file (BlackApps.dat) on my system, plus I can't find Nahimic or any of the 'Sonic" apps listed in the link. It's not just Pusua though. Fusion 360 has [...]

[4 more replies]

closed 2026-09-29 by prusaslicer-bot[bot]