prusa3d/PrusaSlicer #5576: Feature request: Option to set z-offset for the first layer

nnmalex, 2020-12-24:
It would be great if there was an option to change z-offset for the first layer to help with bed adhesion similar to how it’s done in Cura.

rtyr, 2020-12-24:
This is not enough?

[image]

nnmalex, 2020-12-24:
Not really. I have a well calibrated Z offset but some filaments just wouldn't stick to bed well enough and what I do is play with Baby steps when the print is starting just for the first layer.
Cura has an option to alter Z offset only for the first layer and this helps a lot with adhesion keeping the rest of the print high quality.

Kachidoki2807, 2020-12-25:
So Cura squish the very first layer only an then adapt the height of the second layer? Do you have a G-Code with that feature enabled?

I don't see how to squish only the first layer without impacting the next layers because the second one would not be squished enough over the first one if its offset is suddenly raised up.

The combinaison of babystep, first layer height, and elephant foot compensation should be enough to have a perfect first layer. Eventually slicer Z-offset might help, but it is redundant with the babystep. Maybe a custom G-Code in the filament profile to add an offset per filament (there is no override for that) would be more relevant.

nnmalex, 2020-12-25:
Please see attached the sample gcode from Cura. That's a simple cube with no infill. The first layer has Z offset of -0.1mm added to whatever printer settings are.

The point is that with Cura I don't need to fiddle with baby steps, just send to the printer and forget about it whilst with PS I do for some filaments.

[cube.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/5741965/cube.gcode.txt)

Area5142, 2020-12-25:
I'm no expert in g-code, but to me it looks like the nozzle is lowered 0.1 mm before printing first layer:
```
G0 F6000 X89.699 Y89.765 Z0.1 ;adjusted by z offset
G92 Z0.2 ;consider this the original z before offset
```
but not raised again before second layer.
This is exactly the same as setting z-offset in PrusaSlicer (or manually inserting a G92).

Why is there a need for only lowering the first layer, it would create bad bonding between first and second layer?
If it is to get better bed adhesion, there are other solutions, like increase extrusion multiplier for first layer (M221), wider first-layer extrusion width or higher first layer.

nnmalex, 2020-12-25:
Yeah, after looking closer at this G-code I agree with you. That's not just for the first layer. Never mind then.

closed 2020-12-25 by nnmalex