prusa3d/PrusaSlicer #5577: Error (not well-formed (invalid token)) while parsing '3D/3dmodel.model' at line 4

CWStevens, 2020-12-24:
### Version
Version 2.3.0-rc2+win64
(But may have been a **save** issue from 2.3.0-rc1)

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
[image]

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Mk3s

### Behavior
* _Describe the problem_
Attempted to load a recent .3mf file.  Previously saved with RC1.
* _Expected Results_
Load the .3mf file.

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
"Error (not well-formed (invalid token)) while parsing '3D/3dmodel.model' at line 4"
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-12-24:
Thanks, but can you please post the 3mf? Is the crash reproducible with e.g. any previously selected printer profile?

CWStevens, 2020-12-24:
[Evil_Puzzle_-_With_2x6mm_Magnets (Michael & Leah).zip](https://github.com/prusa3d/PrusaSlicer/files/5740619/Evil_Puzzle_-_With_2x6mm_Magnets.Michael.Leah.zip)

lukasmatena, 2020-12-25:
Thanks, reproduced. It is really a saving issue, apparently already present in 2.3.0-alpha1 (since 6e8006524009cd10322cc3418cf4500ceb833978). We newly save the name of the object in the object file metadata, but some characters need escaping in that case (it is an XML). This escaping was missing, so any file containing either `<` or `&` in the filename would be unreadable.

Fixed in branch `lm_xml_escaping`, will be merged after it is reviewed.

commit 6c2ce4b6 (2021-01-04): Escape special xml characters when writing metadata into the model file
This should fix #5577
closed 2021-01-04 by lukasmatena