prusa3d/PrusaSlicer #5578: The configured post-processing script does not exist: Print

iprepiprep, 2020-12-24:
### Version
2.3.0-RC1, 2.3.0-RC2

### Operating system type + version
WIN10-64bit

### 3D printer brand / version + firmware version (if known)
Duet2

### Behavior
* _Expected Results_
gcode is transferred and print starts

* _Actual Results_
When sending gcode to printer (over TCP/IP) I am getting Prusa Slicer Error "The configured post-processing script does not exist: Print". Physical printer is configured and connection test passes.
It worked fine with Beta3.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Independent of a specific object

iprepiprep, 2020-12-24:
[](url)

iprepiprep, 2020-12-24:
[Part Studio 1 - Part 1(5).zip](https://github.com/prusa3d/PrusaSlicer/files/5740349/Part.Studio.1.-.Part.1.5.zip)

FidelCapo, 2021-01-05:
I'm trying to reproduce this issue, but there is no post script in the attached 3mf file.

iprepiprep, 2021-01-05:
I do not any post script files. Also, found some additional symptoms:
- The error occurs also when saving to a file
- It seems to be related to only one of the profiles. I will export my profiles and attach to this ticket.

iprepiprep, 2021-01-05:
See my profiles attached.
The issue is observed with:
- Print: 0.1mm - Quality
- Filament: 3D Solutech PETG Clear - 0.1mm Qualty Optimized
- Printer: T0-Hemera 0.4mm

but is not observed with other profiles.
[
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/5772167/PrusaSlicer_config_bundle.zip)

](url)

foreachthing, 2021-01-06:
If I look at your .ini, there is some information where none should be:
`post_process = ;"Print Setting: 0.1mm - Quality v1.0"`
You cannot have **anything** (nope, not even a "comment") in `Post-processing scripts` if you don't have one.

iprepiprep, 2021-01-06:
Thanks. That solved the problem. I am curious how it got there. The interesting part is that this text does not show up in the UI

foreachthing, 2021-01-06:
> The interesting part is that this text does not show up in the UI.

My guess is, it's because it's a comment.
Maybe you could overwrite it in the UI with some text, then clear it?

_and you may close the issue if it's solved for you ;-)_

iprepiprep, 2021-01-06:
Yes. Typing in the UI and saving the config clears that comment
Thanks!

closed 2021-01-06 by iprepiprep