prusa3d/PrusaSlicer #5581: [Bug] Seemingly working STL is sliced with a gap in the middle

RichiH, 2020-12-25:
### Version
2.3.0-rx2+linux-x64

### Operating system type + version
Debian testing & unstable

### 3D printer brand / version + firmware version (if known)
Prusa Mini +, Firmware 4.2.1

https://www.thingiverse.com/thing:4147318

### Behavior
When slicing https://www.thingiverse.com/thing:4147318 I always get a floating top. STL is fine in other programs. 
[image]

neophyl, 2020-12-26:
The model has 46 non manifold edges, 102 zero faces and 18 thin faces.  Models should be manifold to slice properly.
In fact on the imported model before slicing it you can see a problem area in the top right edge (theres a dark green area)

For future reference if you are using the Windows 10 version you can use the built in fix through Netfabb option to repair files that have slicing issues.  However on linux you would need to use the netfabb online service.  Or fix them like I did in Blender using its built in 3dprint tool.  I'm sure there are other tools that will also repair the file(s) for you.

A comment on the original thingiverse item mentions there are issues with the file.

I have attached the fixed version for you.
[Clip_LBoxx_FW_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/5743719/Clip_LBoxx_FW_fixed.zip)

RichiH, 2020-12-27:
Nice, thank you!

Do I assume correctly that fixing STL is out of for for Prusa slicer?

Can't close the issue through email, need to do so tomorrow.

Sent by mobile; please excuse my brevity.

RichiH, 2020-12-27:
Thanks again, @neophyl !

closed 2020-12-27 by RichiH