prusa3d/PrusaSlicer #5584: Information text on startup logo - Faulty position?

mippen, 2020-12-25:
### Version
2.3.0-rc2+win64

### Operating system type + version
win64

### 3D printer brand / version + firmware version (if known)
Mk3s

### Behavior
Info text on startup seems to be on the wrong position.
"Preparing settings tab" is written over "Version" text.
[https://photos.app.goo.gl/gZWRFbB7HvMbiyUw7](url)

Kachidoki2807, 2020-12-25:
I guess it is not a wrong position of the "Preparing settinds tabs" but a wrong font size of "PrusaSlicer Version [...]".

Your picture:
[image]

What it should be:
[image]

YuSanka, 2020-12-29:
Can you please send me your display settings? I suppose, you use more than one display with different settings.

mippen, 2020-12-30:
I'm running 3840x2160 with 250% zoom on this laptop.

bubnikv, 2021-01-08:
Fixed in PrusaSlicer 2.3.0-rc3.

closed 2021-01-08 by bubnikv