prusa3d/PrusaSlicer #5585: Feature request: Lift Z only on top layer

jbro-oregon, 2020-12-25:
### Version
2.2.0

### Operating system type + version
Windows 7 Professional 64-bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2S FW: 3.2.3

### Behavior
I sometimes use material that tends to string/drool, like Proto-Pasta Matte HTPLA. To alleviate stringing issues, I turn off Lift Z. The extruder then has minimal drooling/stringing during non-print moves over the previous layer. This results in lines across the previously printed areas, but they are on internal layers so it is not an issue. The issue is on the top solid layer, where the lines from non-print moves are very evident.
The current Lift Z settings only allow for a window based on fixed Z-distance settings in mm. I would like the ability to only turn Lift Z on for the very top layer. This would eliminate the "drag" lines, and any stringing would easily be removed in post-processing and not effect the print quality.
I realize that over-extrusion can cause similar issues; I am not over-extruding.

I can cause the desired behavior partially by modifying the Only Lift Z: Above Z: [enter value based on sliced model max Z minus a layer]. But then it must be modified every sliced part. Also, if the part has a stepped top (i.e. flat surfaces at multiple Z heights), then this only helps on the portion that is on the final print layer. It would be nice to have a checkbox for "Only lift Z for top solid infill layers".

This is a new feature request.

erlis, 2023-03-02:
And here we are 2 years after asking for this feature! How hard would it be to implement this? Please consider adding this to the next release!

Dakkaron, 2023-05-30:
Here's more evidence to the matter: https://www.youtube.com/watch?v=4xyevbElGIU

Z-hop seems to be detrimental on all but the top layer.

CogsOz, 2023-06-17:
Adding my +1 to this feature request.

I'm not sure how good the implementation is but SuperSlicer has this option.  It's too far behind PrusaSlicer now and looking like it might be dying a slow death, so I'd love to see this useful feature find its way here.

abortz, 2023-08-04:
And now OrcaSlicer has this feature as well, as of 1.6.4-beta. https://github.com/SoftFever/OrcaSlicer/pull/1562

ramosglauco, 2024-03-19:
There goes 4 long years.... How difficult will it be to implement this improvement? Jesus Christ

gudvinr, 2024-03-20:
@ramosglauco why didn't _you_ sent a PR then, if that isn't that difficult?

JPK-MG, 2024-04-01:
I sure wish that I was a decent programmer, because I'd love to be able to get this feature on board. My workaround is to manually insert the g-code exactly where I need it, but it gets tedious as soon as you want to change any values to your presets - you have to redo all of your g-code.

If I could create a height range modifier (or something similar) and then manually inject my code there... like; Line #202020 Inject: G1 Z2.4 F7200 and then on Line #202025 Inject: G1 Z2.0 F7200 (for a 0.40mm lift) 
... that would be useful on many fronts.

Yes, Cura has better control options for the top surface, but I specifically came back to Prusaslicer for the stable features and printing quality. I'll happily use it, with, or without this feature

Dakkaron, 2026-09-02:
This issue is still relevant.