prusa3d/PrusaSlicer #5586: Prusa Crach shen copy and past.

mullcom, 2020-12-25:
### Version
_Version of PrusaSlicer used goes here_
2.3.0-rc1 +win64
PrusaSlicer-2.3.0-rc1+win64-202012172027
[Huv.zip](https://github.com/prusa3d/PrusaSlicer/files/5742432/Huv.zip)

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Microsoft windows 10 Version 20H2 (OS Build 19042.685)
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.3.0-rc1+win64
Build:     PrusaSlicer-2.3.0-rc1+win64-202012172027

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,302MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.89
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1660 SUPER/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
[... 352 lines cut ...]
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
prusa3d_fw_3_9_1_MK3S - Original Prusa i3 MK3S & MK3S+ - Copy

### Behavior
* Application shuts down with no worning.
* I do copy and past with shortcuts on keybord. of a Generic-Box_ with type Support Blocker

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
Have a lot of features =D Enoying things that can be improve Work time in the application.
Cloude Footprint of the application. If not Prusa self why not a own cloude i can have all my settings in. Like a Android or now windows also start to backup settings.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-12-25:
Thanks. Could you please try with [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2)? I think this might be fixed now (related to #5517).

mullcom, 2020-12-26:
> Thanks. Could you please try with [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2)? I think this might be fixed now (related to #5517).

Yes I can. Need to find the time.

I did some more trying and it seams to start to working and didn't crash. So I thing this happen in a specific step of change.

Anyway. I noticed that change settings for support didn't update and support disappear. To get it back I was needed to change view back and forth to see support again and the new settings.

mullcom, 2020-12-27:
So í have test this in Rc2. no crash has uncured.

But have hard time to get support generate effective. Take a lock at this file i upload so you understand.  

1. Still need to move back and forth to se changes when i change settings in support.
2. Miss calculation of support waste of material. It add to walls on the other but only one on the first.
[Huv2.zip](https://github.com/prusa3d/PrusaSlicer/files/5744909/Huv2.zip)

should i make a new issue on this?

lukasmatena, 2020-12-27:
Thanks, I'll close this issue.

> Still need to move back and forth to se changes when i change settings in support.

If you are sure that it is a bug, then yes. But from your description it seems that you should just enable Background processing in  Configuration->Preferences.

closed 2020-12-27 by lukasmatena