prusa3d/PrusaSlicer #5587: Missing parts in sliced model

MilanNXT, 2020-12-25:
### Version
2.2.0
2.3.0-rc2

### Operating system type + version
Windows 10 20H2
[led-star.zip](https://github.com/prusa3d/PrusaSlicer/files/5742499/led-star.zip)

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
sliced model is missing parts from original STL
STL has no errors. It slices correctly in Simplify3D

below you could download and test problematic STL. both STL generated in openScad.
[stl1](https://musec.visualstudio.com/projects/_git/led-star?path=%2Fled-star-cross.stl)
[stl2](https://musec.visualstudio.com/projects/_git/led-star?path=%2Fled-star-tip.stl)

same error on any printer profile Mini, MK2, MK3

5jvm0u4, 2020-12-26:
same issue here

neophyl, 2020-12-26:
To the original poster.  The Led star tip model does have a couple of issues.

Look at this area for example 
[image]

problem geometry from the faces here.  In exactly the area of that model that doesn't slice.
While the model is manifold and PS should be able to handle it saying there are no issues is incorrect.

As for the Led star cross model, it is perfect.  Not a single vertices out of place.  However after I imported it into Blender to check it over and finding zero wrong with it I exported it again and then loaded it into slicer.  The export from Blender now slices perfectly, even with no changes to the actual model.

This would suggest something strange going on with Openscad's export/generation of the stl format.  A lot of the problem models here and on the Prusa Forum are from openscad though so improving handling of openscads quirks would probably be a good thing as it would cut down on a lot of these 'my model is missing parts' issues.

Here are the edited versions I exported and successfully sliced with in case you want them.
[led-star_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/5743091/led-star_fixed.zip)

MilanNXT, 2020-12-26:
Thank you for prompt review. Let me test to import sand export models through blender. It seems its some common issues with openScad but in past slicer was able to identify and eventually fix the problem or at least drop warning so it could be fixed. What confused me was that simplify3d had no issue with it.

MilanNXT, 2020-12-26:
Thank you for prompt review. Let me test to import sand export models through blender. It seems its some common issues with openScad but in past slicer was able to identify and eventually fix the problem or at least drop warning so it could be fixed. What confused me was that simplify3d had no issue with it. Slicer developer team might consider enhancing some checks for such errors. Thank you again for cheking on the issue promplty.

closed 2020-12-26 by MilanNXT

lukasmatena, 2020-12-26:
Thanks to @neophyl for looking into this issue. It is true that many of the broken models we see reported here come from OpenSCAD. And it is also true that competition slicers sometimes do better job when making sense of them, although I also remember cases where it was the other way around.

@MilanNXT Let me just add that if you're running Windows, you should be easily able to use NetFabb service to fix the model for you (after right clicking the model). I did not check, but I assume that would have fixed the issue too.

neophyl, 2020-12-26:
I didn't check the fix through netfabb option.  Sorry.
I'm currently isolating until the 2nd Jan as the covid app (NHS - UK) has informed me I have been in contact with someone who has Covid.  As such I'm a little bored :)

lukasmatena, 2020-12-26:
> I'm currently isolating until the 2nd Jan as the covid app [...]

I'm sorry to hear that. Anyway, if you chose to use that time to tidy up our issue tracker, be our guest :-)