prusa3d/PrusaSlicer #5588: Machine limits section not visible in advanced settings for custom printer profile

mvadu, 2020-12-26:
### Version
Version 2.3.0-rc2+win64

### 3D printer brand / version + firmware version (if known)
Custom printer profile for Cetus MkII with smoothie board

### Behavior
Defining a new custom printer does not add a machine limit section to settings in advanced mode
[image]

_Is this a new feature request?_
No

But selecting a Prusa printer adds this section, which is useful to modify machine limits to get a proper time estimate.

neophyl, 2020-12-26:
I think it is dependent on what type of gcode flavour you have selected.

rtyr, 2020-12-26:
It is visible only for Marlin g-code flavor.

mvadu, 2020-12-27:
@rtyr is there any issues in exposing this to Smoothie flavor? My current time estimates are quite off, and using this to define the acceleration settings to match what is defined in Smoothie config might probably makes for an accurate `estimated time to print`

bubnikv, 2021-01-05:
If you place the G-codes setting the machine limits manually into your
start G-code, they should be respected.

DivingDuck, 2024-08-08:
I know this is an old issue, but still an existing annoyance. Is there any possibility to enable this for Smoothieware? The firmware was able to support this all the time since beginning.

closed 2026-09-29 by prusaslicer-bot[bot]