prusa3d/PrusaSlicer #5589: Filament diameters must match exactly to enable wiper tower

gnydick, 2020-12-26:
### Version
Version 2.3.0-rc1+win64
Version 2.3.0-rc2+win64

### Operating system type + version
Windows 10 64-bit

### Behavior
trying to use multiple extruders doesn't work if the filaments' diameters do not exactly match. I get "Invalid Data" instead of "Slice" on the button and a warning dialog in the bottom right of the plater view

I was mixing 2 different filaments, one is really thin, so it is set to 1.55mm. the other was 1.75 like normal. So, once I changed the 1.55 to 1.75 it worked. I think you're going to have to have a "class" of filament diameter.

#### Project File (.3MF) where problem occurs

[elastic bowden loop v6.zip](https://github.com/prusa3d/PrusaSlicer/files/5742673/elastic.bowden.loop.v6.zip)

Kachidoki2807, 2020-12-28:
Hi,
It took me a while to reproduce it from scratch, as I often mix a 1.80mm filament with 1.69mm whitout issue.
In fact we have to go beyond **1.6mm to 1.9mm included** to get this error.

About the nozzle size when using multiple extruders, they must **exactly match**.

I'll be curious to know what technical issue is behind this limitation. It is still atypical but more and more popular (since the rise of the toolchangers) to have a dual nozzle setup with for instance, a 0.6mm nozzle for the infill, and a 0.4mm for the rest.

bubnikv, 2021-01-05:
Wipe tower has been implemented initially for the single-extruder mulit-material setup, thus this limitation is a result of the initial implementation. I am not sure whether the limitation still applies.

lukasmatena, 2021-01-05:
> I am not sure whether the limitation still applies.

It does. The wipe tower is still unable to use nozzles of different diameters. The initial implementation made a lot of assumptions about many things and while many of them have been generalized later, this one wasn't.

bubnikv, 2022-01-20:
@Kachidoki2807 

> It is still atypical but more and more popular (since the rise of the toolchangers) to have a dual nozzle setup with for instance, a 0.6mm nozzle for the infill, and a 0.4mm for the rest.

Does anyone use such a combination regularly? While I understand the potential theoretical benefit, how does it look like in reality?

Kachidoki2807, 2022-01-20:
Looking back, I am not sure of the benefit, because the potential saved time is lost during the toolchange.

Since this time, to achieve a similar feature, I often use larger extrusion width for the infill, solid infill or even internal perimeters to speed-up prints, while keeping a 0.4mm nozzle. This does the job as we can reach out pretty easily the limit of the MVS of an E3D V6 hotend without necessarily need to increase the nozzle size. Even if a 0.6mm can help to increase this MVS limit a bit. The idea behind is to reduce the quantity of movements while getting the same amount of material, same wall thickness and so on.
To sum up, we can think about this like the variable layer height feature, but applied to the extrusion width.

I guess this idea is still applicable on some heterogeneous setup, like a V6 + Volcano. But who use that?

I often use the wipe tower, even on my IDEX, associated with silicone anti-oozing blades. This is the best and the most reliable way to have clean toolchanges without dripping stuck on the nozzle or blobs or under-extrusion on the part. Based on thousands of toolchanges I made with this setup, I mean it reach a very impressive 100% success.

In fact, my biggest concerns about the wipe tower are rather:
- Unable to force a purge for each layer (avoid heat creeping and burning the filament when heating at parking for hours)
- Unable to make it hollow (wide footprint with minimal material)
- Unable to have independent towers for each extruder (would help while using materials that don't sticks together)
- Unable to reduce the default purge volume for a printer profile (70mm³ is huge for my usage, 5mm³ is more than enough, but the footprint is too small)
The first point can be workaround-ed by adding and assigning a cube beside the part, for each extruder. The last point needs a manual overwriting after each launch of PS.

This is off-topic here, maybe it deserves a new issue, or update any existing one to share my experience on this.

bubnikv, 2022-01-21:
@Kachidoki2807 Thanks. We will implement a priming tower for our XL, we believe that will cover most of your requests.

bubnikv, 2022-01-21:
> Unable to force a purge for each layer (avoid heat creeping and burning the filament when heating at parking for hours)

Shouldn't the parking cool-down take care of that? Or could the slicer possibly calculate that the extruder will be parked for a very long time and to cool it down more aggressively?

> Unable to have independent towers for each extruder (would help while using materials that don't sticks together)

Do you really need that? Which two materials do you print that do not stick together? I know that PVA does not stick much to other materials, for our single extruder multi-material we implemented in 2.4.0 that the outermost perimeter of a wipe tower is to be extruded with non-soluble material to make the wipe tower stable. Printing two wipe towers means each material will extrude at each layer. 

Do you use some slicer that implements this strategy?

Kachidoki2807, 2022-01-21:
> Shouldn't the parking cool-down take care of that? Or could the slicer possibly calculate that the extruder will be parked for a very long time and to cool it down more aggressively?

This probably should be okay-ish, at least for the heat creeping. For burning the filament, I mean for regular materials it's ok. But for some sensitive material, I am afraid it is doing some annealing and made the filament stuck into the nozzle. Or any kind of chemical reaction. I know the PolyMaker PolySupport to behave weirdly when heated for a while, it becomes completely liquid, and even once, I got the nozzle literally spitting on me (luckily not into my eyes) when leaving the parking, because of the pressure.

> Which two materials do you print that do not stick together?

A lot of couples, PLA/PVA as you mentioned, but also PETG/PLA, PLA/TPU, ABS/SR-30 and many more... Some are more or less sticky. Generally speaking, support filament being breaking-away type are problematic. The main problem is, when using a small footprint tower (few mm³) but tall, it tends to break between layers.

> Do you use some slicer that implements this strategy?

As far as I know, Ideamaker implement this.

[image]

The two cylinders are the wipe tower. It can take a rectangular or a circle shape, and there are several strategies available

[image]

Generally speaking, when I benchmarked various slicers to be able to handle my custom IDEX i3, I found Ideamaker to have a lot of good ideas. But I still have kept PS. :)

closed 2026-09-29 by prusaslicer-bot[bot]