prusa3d/PrusaSlicer #5591: [Feature Request] Variable line widths

stefanpieter, 2020-12-26:
Hello Community 

The variable line widths feature in the Cura Arachne engine demonstrates the advantages of this feature. I believe that needs to be the new standard for slicers as the advantages are clear.

In addition expanding on this feature to have a wider range of line widths would mean one can:

1. reduce the number of perimeters, while retain the perimeter thickness; 
2. reduce printing time;
3. increase part strength 

In effect dynamically changing the effective nozzle size.

Area5142, 2020-12-26:
The Cura Arachene engine is in early alpha and a description can be found [here](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha).

Hooch180, 2021-01-06:
I also would like to request this feature.
On forum it was mentioned by user joules.

> I wish the slicer could calculate a variable extrusion width for "perimeters", within user pre-set limits. Optimized to avoid gap fills and perimeter collapses.
> 
> This would help reduce alot of travel moves for gap fill when a part is made solid using many perimeters, which is probably the best way to achieve high strength. I currently have to finetune the extrusionwidth to say 0.7489, while notoriously matching it with external perimeter width of same presicion, in order to avoid gap filling. Those filler lines add stringing from retractions, time, blobs and underextrusion upon restart. Yeah, compensating with extra material on restart will mess up those travel moves that are above the trigger retraction threshold, while it cant be increased enough to actually make a difference after a long travel move.

townkat, 2021-01-16:
4. and probably the most important, i can't see how you missed this :): increase detail reproduction as smaller widths produce better details/small features (small letters, sharp angles etc.)

LightningShark25, 2021-02-24:
Have admins/debelopers had any thoughts on this?  I have not looked at what cura has done personally but think this is a fantastic jump in slicing capability.

I have been printing parts where at times the fill gaps can be as high as 20% which is painful. I appreciate that sometimes this can be tuned out by adjusting line width manually.. But so far I have found this a trial and error slicing process that is very time consuming.

Keen to know people's thoughts on the pros and cons of going down this path. Thanks

MR2C280, 2021-12-06:
I, too, would very much like this feature (as I spend a lot of time fiddling with perimeter widths to try to reduce gaps).  Prusaslicer V2.4 has added a new parameter to enable / disable gap fill (to replace Gap fill speed=0  #2956 #5999). Perhaps an additional option of "Vary perimeter width to fill gaps" could be added.  Since gaps are, by definition, less than one perimeter width, increasing the perimeters on either side of the gap by <50% shouldn't create any problems (Stefan at CNC Kitchen has shown, you can print up to 200% of nozzle diameter).  However, it might be wise to only vary the width of internal perimeters so the external surfaces are unaffected.

stefanpieter, 2022-04-23:
Cura is officially live with this...

danoveb, 2022-05-02:
I have just played around (no actually printing) slicing some various objects in Cura 5.0 and it's magic. I always make wall thickness in multiple with the nozzle size, but you always get some gap fills in various places, but in Cura they are completely gone! I went from Cura to Prusaslicer mainly because of the UI, but this feature alone might bring me back to Cura.

jonosaur, 2022-05-08:
It is a very interesting dovelopment, the gap fill pieces add so much time on some models.  But  the beta of cura still add time to prints compared to prusaslicer.  If this does come to prusaslicer it will keep it at the front for slicer choice in my very humble opinion 😃

vtjballeng, 2022-05-09:
Another vote for this feature. This is critical for optimally printing parts where strength is critical.

androandra, 2022-05-15:
I vote for this, too.

kbranch, 2022-06-11:
I'd also love to see this, for what it's worth. Extra detail would make some projects much easier.

bubnikv, 2022-06-29:
Ported from Cura to PrusaSlicer 2.5.0-alpha2. Thanks Cura team for keeping Cura open source and for their effort and dedication.

closed 2022-06-29 by bubnikv

GilesBathgate, 2022-07-22:
Where is Arachne?

I was expecting to find this:

[image]

But in Version 2.6.0-alpha0 I don't see that option:

[image]

Neither can I find "Perimeter generator" in the source code with grep 🤷‍♂️

jmd73, 2022-07-22:
@GilesBathgate It is in the branch master_250

krom9ra, 2022-07-22:
@GilesBathgate Version `2.6.0-alpha0` older than `2.5.0` 🤷‍♂️ It has been tagged at May 19 with some special features;
Arachne engine has been supported in `master_250` branch, you could easily test it with [2.5.0-alpha3 builds](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3)

GilesBathgate, 2022-07-22:
@krom9ra I think it's just that 2.6.0-alpha0 is the version stamp put on `master` which is what I was using, and it is 28 days old :( `master_250` seems to be the new development head. Maybe now is a good time to migrate to a `main` branch for "reasons" ;)

GilesBathgate, 2022-07-22:
@jmd73 @krom9ra Just to confirm the option is present when built from the `master_250` branch thanks!

[image]

sisuwest, 2022-08-19:
[image]

Detect thin walls is greyed out. Is this because of this feature?

hejllukas, 2022-08-20:
Yes, "Detect thin walls" can only be used with the Classic perimeter generator because this option is there to resolve some issues that the algorithm of the Classic perimeter generator has.

Arachne works differently, and it doesn't have the issues for "Detect thin walls" is needed. So because of that, the option "Detect thin walls" is greyed out when Arachne is used.