prusa3d/PrusaSlicer #5592: Slicer crashing with empty buildplate

glstott, 2020-12-26:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Home version 1909

### 3D printer brand / version + firmware version (if known)
I have two and experience the issue with both. Prusa MK3S and a Monoprice Mini Delta. Up to date firmware on the former and a custom marlin on the latter. Neither are connected to my device during the crash. 

### Behavior
* Application functions normally, allows me to work and set things up before crashing without any warning or notification and just disappearing. 
* I open the application. If this is my first time opening PrusaSlicer since restarting my PC, I operate as normal, and usually at some point in the process I hear a sound similar to the windows administrator privilege request sound. Then, it crashes and disappears (no consistency in what I'm doing at the time of the crash). After the first crash, each time I open it up, it crashes much more quickly such that the application becomes unusable. For example, while trying to get the version in the about page for this bug report.

Not a new feature request

#### Project File (.3MF) where problem occurs
No problematic files in particular. Usually I don't have anything on the build plate by the time it happens.

lukasmatena, 2020-12-26:
Can you please test it with [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2)? it is quite possible that the issue will be no longer present.

If that version also crashes, could you run it from the command line with `--loglevel=5` argument if that gives any clue? You can also try to run with `--sw-renderer` - if that helps, we will at least know that the problem is OpenGL related (did you try updating your drivers?). Some suspicious crashes were already reported on Win (#4316, #3374, #3177, #2939), this may be related.

glstott, 2020-12-27:
Tested it with 2.3.0-rc2, and that resolved the problem. After one day of lots of active use on the slicer, no unexpected crashes. Thank you for the help!

closed 2020-12-27 by glstott