prusa3d/PrusaSlicer #5593: Inappropriate firmware upgrade warning for i3 MK2.5S printer

riutortmd, 2020-12-26:
### Version
version 2.3.1-rc1+

### Operating system type + version
OSX Catalina

### 3D printer brand / version + firmware version (if known)
MK2.5s with firmware v3.9.1

### Behavior
gcode file triggers an "upgrade firmware to 3.9.2" message on the i3 Mk2.5s printer, however, the latest available firmware for this this printer is 3.9.1 (3.9.2 is for the Mk3 printer, not the 2.5s)

This behavior does not occur with PrusaSlicer 2.2.0+

#### Project File (.3MF) where problem occurs

; generated by PrusaSlicer 2.3.0-rc1+ on 2020-12-26 at 15:44:30 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.45mm
; first layer extrusion width = 0.42mm

M73 P0 R6
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P2000 R1500 T2000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E4.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
;TYPE:Custom
M862.3 P "MK2.5S" ; printer model check
M862.1 P0.4 ; nozzle diameter check
M115 U3.9.2 ; tell printer latest fw version
G90 ; use absolute coordinates

[firmware_upgrade_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/5743303/firmware_upgrade_bug.zip)

Kachidoki2807, 2020-12-26:
Hi,
>the latest available firmware for this this printer is 3.9.1 (3.9.2 is for the Mk3 printer, not the 2.5s)

The Release page for the 3.9.2:
https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.2
It is mentioned that this version is for MK2.5S too and you can even find the binary for this printer.

rtyr, 2020-12-26:
The latest FW for MK2.5S is 3.9.2, it is just not posted on drivers page for some reason, altough MK2.5 MMU versions are available there...

rtyr, 2020-12-27:
Configuration update 1.2.1 reverts MK2.5 FW notification to 3.9.1 for now. Closing.

closed 2020-12-27 by rtyr