prusa3d/PrusaSlicer #5594: Program Crash: ExtraRenderers.cpp when trying to export G-code

n8bot, 2020-12-26:
### Version
Slightly modified latest master branch as at d2b45c282e70ae75e1f99c158103833a667e0e5c
(modifications are all libslic3r related)

### Operating system type + version
Win 7 64

### Behavior
I was modifying a print setup. I had removed 3 of the 4 instances, re-sliced, and exported G-code. As soon as I clicked save after choosing the filename, the program hung in a weird way, then it crashed back into VS with the following:

[image]

#### Project File (.3MF) where problem occurs
Unfortunately I can not share this 3MF file publicly, but if the Prusa team requires it I can provide it privately, to ensure professional privacy.

n8bot, 2020-12-27:
This same crash happened again. It seems to happen when I remove instances by using the undo command. IE I have added instances, then arranged them, then clicked undo until I only had one left. Tried export the gcode, and crash. Actually, this time it crashed right when I clicked export, no option to choose filename.

bubnikv, 2021-01-05:
It would be great if you could come up with a simple reproducible sequence,
ideally with our official builds.

n8bot, 2021-01-05:
I tried everything I could remember to try to reproduce it, but I can not. I'm sorry.

I had a project loaded. I had already sliced the file a few times, and re-sliced. I had added the instances. Arranged. Sliced, went back to the plater view, undo undo undo undo. Sliced. Export->Crash.

I couldn't get it to crash trying again now with the same project and models. Perhaps I had some settings different. Or perhaps it was due to the T-1 command error thing that is now fixed (kinda doubtful there).

bubnikv, 2021-09-30:
> I couldn't get it to crash trying again now with the same project and models. Perhaps I had some settings different. Or perhaps it was due to the T-1 command error thing that is now fixed (kinda doubtful there).

Could not reproduce, closing.

closed 2021-09-30 by bubnikv