prusa3d/PrusaSlicer #5595: MultiExtruder gcode is generated even if only one is configured (M572 codes are wrong)

BEE-KA, 2020-12-26:
### Version
2.3.0-rc2xwin65

### Operating system type + version
Windows 10
### 3D printer brand / version + firmware version (if known)
BLV MGN  Duett Wifi with RepRap
### Behavior
Gcode for multiple Extruder is generated instead of only one.
I use the following configuration for my printer with Mosaic Palette 2 pro:
[image]
The result in Gcode is a M572 S0 D0 code an also 
M572 S0 D1  (for second extruder)   but i do not configured a second.
Can you pleas check this?   
I expect always M572 S0 D0

Greetings,
Frank

rtyr, 2020-12-27:
It is emitted only with the wipe tower enabled. You are using our wipe tower in the combination with the Mosaic Palette? I think it should be disabled in this case, since the Palette will use its own wipe tower.

BEE-KA, 2020-12-27:
Palette does not use its own wipe tower if PrusaSlicer with p2pp is used. 
Older versions of PrusaSlicer it was no issue. (2.2.0)

rtyr, 2020-12-27:
OK. I can confirm it behaved differently in 2.2.0.

FYI @lukasmatena 

Note for us:
p2pp is an independently developed post processing script designed to allow Palette+/2/pro/s users the abiltiy to use Prusaslicer's advanced features nativlely directly from Prusaslicer.

M572 - set pressure advance (similar to M900 in Marlin)

lukasmatena, 2020-12-27:
Thanks, but I don't understand. Can either of you please post a 3MF generated in 2.2.0 that will do something wrong and different in 2.3.0-rc2? As far as I can tell, the M572 is only generated when the wipe tower is generated, which should never happen when only one extruder is used. I'm not saying there is no issue, just that I don't understand it.

lukasmatena, 2020-12-28:
After a private message from @rtyr I understood. The difference between 2.2.0 and 2.3.0-rc2 is a result of fixing #3965, where @marekdovjak rightfully complained that the extruder specification was missing. Which is pretty much the opposite of your suggestion.

I never used P2PP or Palette so I don't know what is the correct behaviour of slicer in your use case. It would be best if someone familiar with how the firmware works and what it does and does not do told us... @tomvandeneede or @ardenpm might know something maybe?

Should the emitted gcode depend on whether "Single extruder MM" is selected?

tomvandeneede, 2020-12-28:
Will add code to P2PP to format all for D0

elif commandNum == 572:
    for index in range(1, len(v.filament_count)):
        g[gcode.OTHER] = g[gcode.OTHER].replace("D{}".format(index), "D0”)

this should correct the unwanted behaviour in p2pp

Version 6.2.1 will get this functionality

lukasmatena, 2020-12-28:
@tomvandeneede Thanks a lot for the prompt reply. Do I understand correctly that current (2.3.0-rc2) PrusaSlicer behaviour is correct and the specific use case of @BEE-KA should be taken care of by your post processor? That is of course the best news for us, I just want to make sure we're not working around some incorrect behaviour of PS.

tomvandeneede, 2020-12-28:
The processing in prusa is correct.

P2PP did not foresee in M572 for now but I added the statement to move all M527 commands to D0 to match the T0 at the start of the print.

lukasmatena, 2020-12-28:
@tomvandeneede Thanks. I'll close the issue then.

closed 2020-12-28 by lukasmatena

BEE-KA, 2021-01-05:
Thank you all for your support :-)