prusa3d/PrusaSlicer #5596: Paint-on supports don't work on MMU2 components

blillesand, 2020-12-27:
### Version
2.3 RC 2

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK3s w MMU2s

### Behavior
Paint on supports do not paint on components of a MMU2 model (only paint on the base model/stl).

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

I don't know how to upload the file.

lukasmatena, 2021-01-05:
Can you please be more specific and ideally provide a 3MF that would demonstrate the issue? Uploading the file is simple, just drag and drop it into the text box where you write your comment. A 3MF has to be zipped first so github does not reject it.

I found no issues during my tests, both painting and support generation worked fine for me. Maybe there is a case where it doesn't, but I did not see it. Thanks.

bubnikv, 2021-09-22:
No response, closing.

closed 2021-09-22 by bubnikv