prusa3d/PrusaSlicer #5598: Bug: Ironing is ignored when slicing if infill patterns are changed

mordhau5, 2020-12-27:
### Version
2.3.0 beta2+win64

### Operating system type + version
Windows 10  10.0.19041 Build 19041

### 3D printer brand / version + firmware version (if known)
Custom/Klipper v0.8.0-843-gf8cb1e30

### Behavior
When slicing, the application ignores the ironing setting and creates gcode without ironing moves if the infill paterns are changed. This problem persists until the ironing option is disabled, sliced, re-enabled, and sliced again. Simply re-slicing is not enough to trigger ironing again.

Changing the infill pattern to any different option in either Top or Bottom pattern triggers this. Changing pattern back to the original and re-slicing does not fix. Was not able to trigger the bug with other settings in the Infill sub-menu or other sub-menus so far. 

Visual Steps:
1) [image]
2) [image]
3) [image]
4) [image]

#### Project File (.3MF) where problem occurs
[PolarBear Parts.zip](https://github.com/prusa3d/PrusaSlicer/files/5744068/PolarBear.Parts.zip)

lukasmatena, 2020-12-27:
Seems like a duplicate of #5410. Can you please retest in 2.3.0-rc2? Thanks.

mordhau5, 2020-12-27:
Ah I checked open issues but forgot to check closed ones. This issue is indeed fixed in rc2

closed 2020-12-27 by lukasmatena