prusa3d/PrusaSlicer #5600: RepRapFirmware Flavor uses wrong M-Code to switch off the fan (M107)

natthapolvanasrivilai, 2020-12-27:
Version
PrusaSlicer 2.3RC2

Operating system type + version
Windows 10

3D printer brand / version + firmware version (if known)
Duet Wifi 2 - RepRapFirmware 3.2RC2

Behavior
As per Duet3D reference, **M107 has been deprecated**. **M106 S0** should be used instead.
When selecting RepRapFirmware as G-Code flavor, the output uses M107 to switch off the fan.
RepRapFirmware should output M106 S0 instead to properly working.

rtyr, 2022-07-12:
> should output M106 S0 instead to properly working.

Is there any actual problem with M107 command? M107 works normally in the latest RepRapFirmware.

closed 2022-09-01 by natthapolvanasrivilai

rsobik, 2023-01-16:
So, since there has been comments on the PR I'd like to point out an issue there is with the current implementation, reported by @Surgikill:

> This issue still persists in prusaslicer 2.5.0, and is causing issues with a duet3 with duet toolboard1LC. The inclusion of M107 in the print file causes issues with the hotend cooling fan, and lowers the RPM of said fan when the file is run. Running the same file with M107 commented out causes no RPM changes of the hotend cooling fan when the file is run.

rtyr, 2023-01-16:
This issue was about `M107` vs `M106 S0`, I believe what @Surgikill reported is different issue.

rsobik, 2023-01-16:
But isn't PR #7537 exactly fixing that: Not issuing M107 anymore, because it is deprecated (and using the correct gcode instead)?

rtyr, 2023-01-16:
> Running the same file with M107 commented out causes no RPM changes

Commenting M107 out is imho not the same as replacing it with M106 S0. Is there any actual difference between using M107 and M106 S0? It behaved the same when I tried it on the printer with Duet 2.