prusa3d/PrusaSlicer #5603: Feature Request - Parameter/Settings Guide

bonjipoo, 2020-12-27:
A settings guide like the one below for Cura would be a great addition.

[https://github.com/Ghostkeeper/SettingsGuide](url)

243k downloads for the Cura version

As more and more settings are made available for Prusaslicer and the text descriptions are sometime not clear for beginners without graphical examples, a guide to parameters would be a handy addition.

neophyl, 2020-12-27:
The only resource I am familiar besides the release notes is on the Knowledge Base here https://help.prusa3d.com/en/category/prusaslicer_204  

The new 2.3 has url links next to some settings for various settings which I think will be expanded upon as time permits.

bonjipoo, 2020-12-27:
I see so this might already be in the cooking pot. 

Prusaslicer 2.3rc2 is fantastic if you ask me.

rtyr, 2021-02-26:
This is what the Knowledge Base is for. It is being expanded continually and as neophyl mentioned, we implemented direct URL links directly to PrusaSlicer.

Closing.

closed 2021-02-26 by rtyr