prusa3d/PrusaSlicer #5604: PrusaSlicer cannot slice this model correctly

damienmg, 2020-12-27:
### Version
_2.2.0_ and _2.3.0rc2_

### Operating system type + version
Linux Ubuntu 18.04.4 LTS

### 3D printer brand / version + firmware version (if known)

Highly modified Creality Ender 3 with SKR 1.3 and Marlin 2.0-bugfix (last fw update one month ago). This is a slicing issue so it does not really matter

### Behavior

* _Describe the problem_

The sliced model does not match the 3d model. The bottom hole is filed, see screenshot. Rendering on the side trigger a rendering errors. Rendering those works 

* _Steps needed to reproduce the problem_

Import the attached model, slice it. For the second one, lay it down on the face, use support and slice it.

* _Expected Results_

Rendered results for the first version with a hole at this place, for the second version with a full body

* _Actual Results_

See screenshots :)

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs

[video_projector_holder_base_top_2.zip](https://github.com/prusa3d/PrusaSlicer/files/5745012/video_projector_holder_base_top_2.zip)

[image]
[image]

tkircher, 2020-12-27:
Opening the model in Blender, I see some significant problems with it. Cleaning up the model and re-exporting it solves the problem.

damienmg, 2020-12-27:
I just opened it with blender and could not find any problems, neither in MeshLab. Why is PrusaSlicer failing silently in one case and totally in the other case? Why Cura has no trouble slicing it?

The model is generated by OpenSCAD with the attached file 
[video_projector_holder.zip](https://github.com/prusa3d/PrusaSlicer/files/5745081/video_projector_holder.zip).

tkircher, 2020-12-27:
PrusaSlicer is okay at repairing broken models, but of course that isn't what it's designed to do. It's more than fair to assume that imported models do not have any degeneracy or non-manifoldness, so any issues where that is the case shouldn't be the slicer's problem. The original model you posted had 329 degenerate triangles and 4 self-intersections.

The issue I see with your OpenSCAD code is your abuse of the hull() function. But that's not a Prusa issue.

damienmg, 2020-12-27:
Indeed, reducing the number of hull remove the issue. This is weird I guess this is an openscad issue then. Thanks for helping there, I'll file an issue to openscad then.

It's interesting that Cura sliced it correctly though but I guess that's depends on the slicing algorithm whether they are impacted by the issue or not.

closed 2020-12-27 by damienmg

tkircher, 2020-12-27:
Try this:

```
module BallPlate() {
    dim = [abs(FRONT_SCREW_POSITION_TO_BACK_SCREW.x) + 20,
           abs(FRONT_SCREW_POSITION_TO_BACK_SCREW.y) + 20,
           SCREW_DEPTH - SCREW_DEPTH_IN];

    difference() {
        translate(FRONT_SCREW_POSITION_TO_BACK_SCREW / 2) {
            // The mount ball
            translate([0, 0, 20])
              Ball(r = RADIUS);

            translate([0, 0, -0.1]) {
              // The main ball spindle
              cylinder(d = RADIUS, h = 20, $fn = 80);

              // The spindle base chamfer
              cylinder(d1 = RADIUS + 4,
                       d2 = RADIUS, h = 2, $fn = 80);
            }

            // The base block
            difference() {
              translate([0, 0, -dim.z / 2])
                cube(dim, center = true);

              // Edge chamfers
              for(i = [0 : 1])
              for(j = [0 : 1]) {
                mirror([i, 0, 0])
                mirror([0, j, 0])
                translate([-dim.x / 2, -dim.y / 2, -dim.z - 1])
                rotate([0, 0, 45])
                  translate([-(4 / sqrt(2)) / 2, -10 / 2, 0])
                    cube([(4 / sqrt(2)), 10, dim.z + 2]);
              }
            }
        }

        // Base countersink
        translate([0, 0, -dim.z - 0.1])
          cylinder(d1 = 15, d2 = 12, h = 3.8, $fn = 120);

        // Mount holes
        translate([0, 0, -dim.z - 1]) {
            cylinder(d = SCREW_DIAMETER, h = dim.z + 2, $fn = 80);

            translate([0, 0, dim.z])
              cylinder(d1 = SCREW_DIAMETER,
                       d2 = SCREW_DIAMETER + 2,
                        h = 1 + 0.1, $fn = 80);
        }

        translate([0, 0, -dim.z - 1]
                    + FRONT_SCREW_POSITION_TO_BACK_SCREW) {
            cylinder(d = FRONT_SCREW_DIAMETER, h = dim.z + 2, $fn = 80);

            translate([0, 0, dim.z])
              cylinder(d1 = FRONT_SCREW_DIAMETER,
                       d2 = FRONT_SCREW_DIAMETER + 2,
                        h = 1 + 0.1, $fn = 80);
        }            
    }
}
```

damienmg, 2020-12-27:
Thanks yes that's basically what I have done.

Thanks a lot for your help. i have seen a bug on Openscad Linux that explains this problem.

lukasmatena, 2021-01-04:
>  i have seen a bug on Openscad Linux that explains this problem.

Can you please reference it here?