prusa3d/PrusaSlicer #5605: Feature request: Add ability to set where an object is placed on the bed

JeffB42, 2020-12-27:
I typically print small to medium sized object, and they don't take up the entire area of the bed.  When I import an object, the object is always placed at the center of the bed.  To reduce wear in the same spot, I'd like the following feature to be added to PrusaSlicer:

- In Printer Settings, allow the user to choose from a 3x3 grid where the default location of imported objects will be placed.
- Then, when an object is imported, the object will be placed in that location (of the virtual grid).  If the object is too large to fit in the pre-designated location, it will be moved towards the center until it fits on the bed.

I know that (presently, in PrusaSlicer) I can move the object after it has been imported, but this would be a nice feature to have, so that the imported object isn't always imported into the same location.

_Is this a new feature request?_
yes

JeffB42, 2026-09-06:
This FR is still topical and relevant.  Please don't auto close this ticket, and implement this feature.

Jan-Soustruznik, 2026-10-01:
Closed as duplicate of #3866