prusa3d/PrusaSlicer #5606: Bug: Settings removed for merged groups do not trigger an "un-slicing" event

mordhau5, 2020-12-27:
### Version
Version 2.3.0-rc2+win64

### Operating system type + version
10.0.19041 Build 19041

### 3D printer brand / version + firmware version (if known)
Custom/ Klipper

### Behavior
* _Describe the problem_
When changing settings applied to a merged group of objects, the objects remain sliced and the "Export G-code" button remains instead of the "Slice" button. Triggering any other change triggers the event and then object slices as expected.

* _Steps needed to reproduce the problem_
1.  have objects on plate, in a merged group, and sliced [image]
2. add a setting for a merged group and slice
I would expect removing settings for the group would cause the sub-objects to re-inherit the print-level settings and require a re-slicing. 
a) [image] 
b) Global Setting: [image]
3. Remove the merged group level setting 
[image]

Another detail that maybe pertains to the intended behavior rather than a bug, but merging an object with specific settings into a group grants all other objects already in merged group the individual object's settings. I would have expected the behavior would leave the individual object's settings at the object level within the merged group. Let me know if you want this in a separate issue for conversation, but this is the behavior that triggered me to realize that the above bug exists.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
[PolarBear Parts.zip](https://github.com/prusa3d/PrusaSlicer/files/5745077/PolarBear.Parts.zip)

RichiH, 2020-12-28:
I have been unable to cleanly reproduce this for a bug report, but I can confirm that the general problem is present in both 2.2.0 and 2.3.0 RC2.

lukasmatena, 2021-01-04:
Should be fixed in `master` by 1d25630 for 2.3.0 final. Closing, thanks.

closed 2021-01-04 by lukasmatena