prusa3d/PrusaSlicer #5609: [Feature Request] Improved seam control

stefanpieter, 2020-12-27:
The seam placement / hiding methods are still a weakness that is letting down a great slicer. This is one area that Cura still produces cleaner prints.

It would be so amazing if seams could be improved :-) so that I only need Prusaslicer in my life.

lukasmatena, 2020-12-27:
> It would be so amazing if seams could be improved :-)

Agreed. It would also be amazing if you told us what version of PrusaSlicer you are unhappy with and what exactly you think should be improved. Otherwise there is not much to discuss. Every part of slicer could be improved.
PrusaSlicer 2.3.0 introduces paint on seam enforcers and blockers. Have you tried it out? There is definitely room to improve, but the discussion would have to be more specific.

stefanpieter, 2020-12-28:
It is difficult to say what to change or add to the feature as this this would be a developer level conversation but... I can clearly see the difference between Cura (4.8) and Prusaslicer (2.3.-rc2) print results and print previews.

https://user-images.githubusercontent.com/65555837/103200228-b7e65100-48e4-11eb-9af3-436e240fbd45.mp4

If I had to equate it to a difference in how the seam is executed it seems like it relates to Cura "wiping" into the print which results in a much cleaner outer appearance.

fscheps, 2020-12-28:
> Agreed. It would also be amazing if you told us what version of PrusaSlicer you are unhappy with and what exactly you think should be improved. Otherwise there is not much to discuss. Every part of [...]

The possiblity of drawing straight lines without having to rely on our "pulse" would be HIGHLY appreciated.

TheDeadPrinter, 2021-01-09:
> Agreed. It would also be amazing if you told us what version of PrusaSlicer you are unhappy with and what exactly you think should be improved. Otherwise there is not much to discuss. Every part of [...]

Yes, I agree with you, for example when I place an object with a flat area at the back a cube or rectangle shape for example, the seam is just placed in the middle at the back! When I set the seam position to the back. There should at least be an option to set the seam to an user specific position like in cura (back, back right, back left, left, right, front left, front right). I don't really want to paint a seam on a complex model. And it is also really hard to draw a straight line, as mention here already. Setting the seam to aligned, doen't help. Yeah, the seam is mostly aligned but nor a the same place!
[image]
[image]
[image]

colatuttorf, 2021-03-12:
I have this same request and can elaborate a bit more on what I'd like to see for some seam improvements (that other slicers have implemented to great effect, and are also sticking points for my switching over to PrusaSlicer full time)

- Seam Aligned with x,y coordinates.  This is the big one really.  For example, in other slicers you can select seam aligned, and then either select from back left, back right, front left, front right, center, etc, or the real valuable option here, is to input your own x,y coordinates for the seam. ie; 0,100 which would give a seam as close as possible to the coordinates x=0, y=100

This can be improved even further than other slicers by taking an approach similar to:
- Seam Alignment Vector: Input x,y coordinates and then the slicer uses them to calculates a vector from the parts centroid to the specified x,y coordinates, and uses that vector to place the seam in the various perimeters of the part, as shown in the sample image below

[image]

jschuh, 2021-12-08:
@colatuttorf, would PR #6727 fit your needs (direction, but no magnitude)?

closed 2026-09-29 by prusaslicer-bot[bot]