prusa3d/PrusaSlicer #5611: line numbers are wrong

3Dietrich, 2020-12-28:
### Version
Version 2.3.0-rc2+
Build:     PrusaSlicer-2.3.0-rc2+-202012221629

### Operating system type + version
Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Catalina Version 10.15.7 (Build 19H2)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 INTEL-14.7.8
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Plus Graphics OpenGL Engine
GLSL version: 1.20
Installed extensions:

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
#### _Describe the problem_
I want to change the g-code (move M600 to a better position) and 
and I thought I can find the right line with Prusa Slicer or G-Code Viewer, but the lines are completely wrong.
Slicer shows g-code print start at line 9:
[image]

but actually it starts at line 222:
[image]

But I couldn't find a way to calculate other line numbers either, not even with offset 213 (222-9) or so... maybe it's me or each layer gets its own offset...?

#### _Steps needed to reproduce the problem_
to reproduce I need the reson for the error, and **I guess** it is because gcode thumbnails. I have this line
[image]
and it works great with octoprint.
My problem is, that I added this feature by hand before 2.3 and tweaked the PrusaResearch.ini and now in alpha, beta and rc several several times.. and I don't know if it's correct, but it works without issues.

Anyway, I can not remove the line ('16x16, 220x124'), slicer won't accept a blank line. So I can't test it as it is without thumbnails...

#### _Expected Results_
it should show the correct line number
#### _Actual Results_
it shows wrong line number
#### Project File (.3MF) where problem occurs
contains 3mf, gcode and PrusaResearch.ini
[slicer shows wrong linenumbers.zip](https://github.com/prusa3d/PrusaSlicer/files/5747522/slicer.shows.wrong.linenumbers.zip)

Kachidoki2807, 2020-12-28:
Hello,
There is a misunderstanding of the meaning of the "line" you circled. There is no relation between the line number into the G-Code file and any of the rulers. The only thing you can look at is the layer number.

In your example, the first layer start to print when you find ";Z:0.4" into the G-Code (line 246).
If you place an M600 here, then it will be placed after the "before layer change" and "after layer change" custom G-Code, and in your example it will be somewhere at line 258.

The "line" you circled are the segments of each individual tool related G-Code, like a move or an extruded line. There is not link with the G-Code line number.

You can use the Custom G-Code sections to add a specific comment as a tag to easily find it into the G-Code file.

3Dietrich, 2020-12-28:
oh.. ok. ..but.. even in the "g-code viewer" app ..there is no g-code line view? I need this quite often (used https://gcode.ws until now) to set e.g. M600 and similiar to the better/correct position.
In fact, I'm just wondering what the display of the
>  segments of each individual tool related G-Code

 is for. Wouldn't lines be more useful?

Kachidoki2807, 2020-12-28:
Both Prusa G-Code Viewer and the bottom ruler are quite new, maybe it is a good improvement suggestion to add a link with the G-Code in text format.
The bottom ruler is used to see how a layer is printed, in which order, or the travel moves over the layer etc... I can help to  anticipate the behavior at some critical areas. But I agree that its numbers have no meaning as is.

I am curious, what is not correctly handled by this function for your use case?

[image]

It is intended to be placed between two layers, why do you want to move it elsewhere?

3Dietrich, 2020-12-28:
> why do you want to move it elsewhere?

A (not really important): the point where M600 is entered is the **last point before** it moves to the new layer/position. 
To be very precise (for example, to avoid weariness when restarting), I set M600 one line later:
[image]

B (more often): color changing is often important to not change it between layers but at the right position in between. 
I explain somewhat stumblingly here in this video:
https://user-images.githubusercontent.com/4608796/103242539-d2591280-4956-11eb-9f6c-f2d034a655ff.mp4

Area5142, 2020-12-29:
@3Dietrich color change in mid layer is related to this feature request #5195.
I would like that too ;-)

3Dietrich, 2020-12-29:
My bug is a bit jumbled.
- Originally: the circled number ("9") should, in my opinion, indicate the line so that you can specifically change things in the g-code via text editor.
- it turns out, it's about color change and I want to move the position in G-Code in external text editor what caused me ot this "bug"report
- but the focus is now more like **"let me 'Add G-code'**(\*) **anywhere _between_ layers"**. 
(*)where "G-code" means "M600" or "M601" or "Custom G-code"

If this could be done directly in Slicer that would be great!

rtyr, 2021-02-26:
The option to show gcode line number on horizontal slider was added. It will be included in the next PrusaSlicer release.

As for the last point, it was already requested in different issues (#5195).

Closing.

closed 2021-02-26 by rtyr