prusa3d/PrusaSlicer #5614: No 3d view of models

angel6700, 2020-12-28:
### Version
Stable 2.2.0 and freshly compiled 2.3.0-rc2

### Operating system type + version
Linux (archlinux)

PrusaSlicer
Version:   2.2.0
Build:     PrusaSlicer-2.2.0+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.9.14-arch1-1 x86_64
Total RAM size [MB]: 4,022MB
OpenGL installation
GL version:   2.1 Mesa 20.3.1
Vendor:       Intel Open Source Technology Center
Renderer:     Mesa DRI Mobile Intel® GM45 Express Chipset (CTG)
GLSL version: 1.20
Installed extensions:

GL_3DFX_texture_compression_FXT1
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_EGL_image_storage
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_demote_to_helper_invocation
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_framebuffer_fetch_non_coherent
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robustness
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

### 3D printer brand / version + firmware version (if known)

### Behavior
* _Describe the problem_
  After loading a STL model, the platter view show only the white selection of the surrounding box, but not the model.
Same apply after clicking in "slice now". It finish but do shows only an empty platter.

* _Steps needed to reproduce the problem_
Run prusa-slicer
Load a stl model

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
 3D view of the model
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

I  now my computer is old and somehow limited, but I think it should work.
I think it is related to opengl, because if I run :
`export LIBGL_ALWAYS_SOFTWARE=1 `
before running prusa-slicer, then it works as expected.

This is probably just a missing library of driver in my computer, but I have no idea what, because the opengl environment seems correctly installed.

Thank you.

angel6700, 2021-01-09:
Thank you,

You are right. It's something with my setup.
It is just because I'm only having troubles with PrusaSlicer, all opengl tests are running fine.

Running with --loglevel 5 do not show any error

`[2021-01-09 19:53:57.932639] [0x00007fd1611d0d80] [info]    Checking if indices need to be installed from resources...
[2021-01-09 19:53:58.323667] [0x00007fd1611d0d80] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 68MB; Shared memory: 51MB; Private memory: 17MB; Peak memory usage: 68MB
[2021-01-09 19:54:01.298146] [0x00007fd1611d0d80] [info]    Checking for cached configuration updates...
[2021-01-09 19:54:01.298531] [0x00007fd1611d0d80] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2021-01-09 19:54:01.298853] [0x00007fd1611d0d80] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2021-01-09 19:54:01.299719] [0x00007fd1611d0d80] [info]    Confing bundle not installed for vendor Creality, skipping: 
[2021-01-09 19:54:01.299892] [0x00007fd1611d0d80] [info]    Confing bundle not installed for vendor PrusaResearch, skipping: 
[2021-01-09 19:54:01.299975] [0x00007fd1611d0d80] [info]    No configuration updates available.
[2021-01-09 19:54:01.362213] [0x00007fd14e76b700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
[2021-01-09 19:54:01.683223] [0x00007fd14e76b700] [info]    Got PrusaSlicer online version: `2.2.0`. Sending to GUI thread...
[2021-01-09 19:54:01.686723] [0x00007fd14e76b700] [info]    Syncing configuration cache
[2021-01-09 19:54:01.686781] [0x00007fd14e76b700] [warning] No such vendor: BIBO
[2021-01-09 19:54:01.687020] [0x00007fd14e76b700] [warning] No such vendor: LulzBot
[2021-01-09 19:54:01.687127] [0x00007fd14e76b700] [warning] No such vendor: Creality
[2021-01-09 19:54:01.687232] [0x00007fd14e76b700] [warning] No such vendor: PrusaResearch
[2021-01-09 19:54:05.121019] [0x00007fd1611d0d80] [info]    Undo / Redo snapshot taken: Import Object: LEDHolder15mm_27mm_15mm.stl, Undo / Redo stack memory: 0MB Resident memory: 116MB; Shared memory: 68MB; Private memory: 48MB; Peak memory usage: 116MB
[2021-01-09 19:54:05.288223] [0x00007fd1611d0d80] [debug]   TriangleMesh::repair() started
[2021-01-09 19:54:05.305500] [0x00007fd1611d0d80] [debug]   TriangleMesh::repair() finished
[2021-01-09 19:54:05.355620] [0x00007fd1611d0d80] [debug]   TriangleMesh::repair() started
[2021-01-09 19:54:05.360986] [0x00007fd1611d0d80] [debug]   TriangleMesh::repair() finished
`
I must say I have ArchLinux installed in this computer and the problem appeared after this recent format & reinstall.

Thank you anyway.

bubnikv, 2021-09-25:
The unfortunate thing about OpenGL API is there is no way to find out the RAM size available to the graphics card.
Sorry, we cannot help.

closed 2021-09-25 by bubnikv