prusa3d/PrusaSlicer #5616: Skirt problem

MaximSharkov, 2020-12-28:
### Version
PrusaSlicer 2.3.0 rc1

### Operating system type + version
Win10 64 2004

### 3D printer brand / version + firmware version (if known)
DIY H-BOT

### Behavior
Skirt do not touch model on preview and in real print.

#### Project File (.3MF) where problem occurs

[image]

foreachthing, 2020-12-28:
Use brim value >0 mm (4 mm, i. e. ) ;-)

MaximSharkov, 2020-12-29:
i know what is brim and skirt, but not work... skirt always print with gap, even if "elephant foot compensation" = 0

[image]
[image]

if distance of skirt from object = 0 it MUST touch model, otherwise this parameter makes no sense

foreachthing, 2020-12-29:
Well, then I don't fully understand your issue.
I understand, that zero should be zero. But maybe it's not zero for a reason: Brim!
Maybe you could have a 2 mm brim and then a skirt 1.6 mm (difference is your default extrusion width) from the object?!
Please explain what you want to achieve.

MaximSharkov, 2020-12-30:
yes, you understand correctly, zero should be zero, moreover hint tells us about the same
[image]

closed 2020-12-30 by MaximSharkov

MasterCATZ, 2021-01-23:
unsure why this was closed , having same issue , skirt and brim are now always separated , they are not being attached to anything and distance is set as 0 PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage

MasterCATZ, 2021-01-23:
skirt and brim 
[image]
[image]

ciapamusche, 2021-02-27:
Also I have same problem . The skirt don't touch the model. 

Never solved the situation.

I have last version of prusadlicer_

Measure 0 has no sense. I have to use another slicer to print warping materials like nylon.

kainzuu, 2024-03-23:
This is still an issue and I have a use case for why it should be re-opened and fixed.  I have a model with a square base with some dovetail cutouts on the bottom.  I want to basically have a square brim that doesn't go into these dovetails while still holding down the edges of the model as cleaning up the brim in the dovetails is a hassle.  As mentioned in previous comments it is inexplicable that this is the behavior when the tooltip says the distance should be zero.  As you can see in the pictures the brim is closer than the skirt.  
[image]
[image]