prusa3d/PrusaSlicer #5618: MMU2 Single filament start code causes giant blob on nozzle before purge.

FEsmondeWhite, 2020-12-28:
Please see extensive reproduction of this issue in the Prusa Firmware Github issue: [https://github.com/prusa3d/Prusa-Firmware/issues/2069](https://github.com/prusa3d/Prusa-Firmware/issues/2069)

### Version
PrusaSlicer-2.3.0-rc1+win64 + all older

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S
3_9_2_MK3S_1_0_6_MMU2S

### Behavior
In single-filament mode, a giant blob is extruded before the purge lines are made.
See detail in linked GitHub issue:  [https://github.com/prusa3d/Prusa-Firmware/issues/2069](https://github.com/prusa3d/Prusa-Firmware/issues/2069)

This is a bugfix, not a new feature request.

#### Project File (.3MF) where problem occurs
[MMAR-Holder-MMU.zip](https://github.com/prusa3d/PrusaSlicer/files/5749071/MMAR-Holder-MMU.zip)
(this is project file for an [AWESOME spool holder](https://www.prusaprinters.org/prints/3729-multiple-mechanism-auto-rewind-spool-holder/comments))

FEsmondeWhite, 2020-12-28:
By the way, this seems to be an issue lots of other people have run into:
[https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/how-do-i-turn-off-that-giant-blob-production-at-the-start-of-the-print-prior-to-the-purge/](https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/how-do-i-turn-off-that-giant-blob-production-at-the-start-of-the-print-prior-to-the-purge/)

Note the solution: don't slice in Single mode. Single filament slicing is currently broken.

rtyr, 2020-12-29:
This is firmware issue not related to start gcode.

FEsmondeWhite, 2020-12-29:
Since it is a firmware issue, could you prompt someone working on the firmware to look at the linked issue report in the Prusa firmware tracker?

Thank you & happy new year!
Francis

FEsmondeWhite, 2020-12-29:
Unresolved previously reported firmware issue. Closing issue in PrusaSlicer tracker.

Thanks!

closed 2020-12-29 by FEsmondeWhite