prusa3d/PrusaSlicer #5619: Copying Item with Printable Property

louwheesy, 2020-12-29:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Not Relevant

### Behavior
When copying an item that has the "printable" property (The eye icons in the part list) disabled, the status copies and moves out of the print area when slicing. This is fine. However, the slicer does not fully darken it and thinks that it is still a printable part, causing an error.
To replicate this, simply toggle the printable property on a part, copy / paste it, then use the arrange command.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]

bubnikv, 2021-01-05:
Do you have the issue with PrusaSlicer 2.3.0-rc2?
https://github.com/prusa3d/PrusaSlicer/releases

louwheesy, 2021-01-17:
Sorry for the late reply. I just tried it in PrusaSlicer 2.3.0 (clean install) and it does the same thing

rtyr, 2021-03-01:
Fixed in master. Closing.

closed 2021-03-01 by rtyr