prusa3d/PrusaSlicer #5622: Not possible to adjust bridge flow ratio depending on layer height. I have to manually edit gcode to achieve what I want. For best printing results I set flow ratio for first layer to 1.18, second layer 1.1 and all following layers to 0.97. But in Prusa Slicer there is not possible to change it regarding to layer. Please add this feature.

edgars-dev, 2020-12-29:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

th33xitus, 2020-12-29:
First of all you shouldn't type your whole issue in the title. Thats what the field below is for with the template you can see there.

A function like this is already in place if i understand you correct.
With the help of height range modifier you are able to alter the bridge flow ratio.
[image]

edgars-dev, 2020-12-29:
I am sorry. Will use the field in the future. 
Oh yes, you are right! Just checked and it is there. However, there are few issues with it.
1) I need to set those every time I add new object for printing. That beats the purpose of the setting, because it actually takes longer than editing gcode. It should be something that I can adjust once in general settings and it appies for all of prints.
2) This setting asks me range and layer height. That most of the times differ for the first layer which brings confusion because, if I set there that layer height is 0.2 but my first layer is 0.3, what would it do? (Retoric question just to point out the confusion)
For this particular setting (Bridge flow ratio) it would be more useful if it would just let change the flow ratio depending on layer (1,2, above) without height consideration and it would be a generic setting that applies to all prints.

Thank you for your fast reply!
Kind regards

th33xitus, 2020-12-29:
Let me ask something.
Why do you even need a different bridge flow ratio already applied at layer one, then change it again on layer two and after that change it again for all other following layers? Can you give an example of a print where you would apply this method? Because i really struggle to find a real world application where i already need to bridge on the first layer...

edgars-dev, 2020-12-29:
Well I don't know why, but when I printed with just "Generic PLA" preset, the prints came out with bottom layer strikingly transparent. I could see the second layer through it easly. But if I just set higher flow ratio then the print is worse brecause of excess material 
Besides it visual looks it also makes part attachment to bed worse and they tend to peel of much more often.
[image]

Yep, the setting that is available in Slicer GUI under "Print Settings", then "Advanced" on the left side and then "Bridge flow ratio" input field is the one that changes M221 setting in gcode.
Yes, that would be awesome if I could just code it myself. Your code looks exactly what I need. Where should I insert it?

th33xitus, 2020-12-29:
> Where should I insert it?
Printer Settings -> Custom G-code and then either in the `Before layer change G-code` or `After layer change G-code` input fields. Personally i would just put it into the `After layer change G-code`.

edgars-dev, 2020-12-29:
Exactly what I needed. Thank you very much. Have a nice day!

rtyr, 2020-12-29:
Can we close this issue?

edgars-dev, 2020-12-29:
Seems so

closed 2020-12-29 by edgars-dev