prusa3d/PrusaSlicer #5623: MMU2S not asking for filament number

alienryes, 2020-12-29:
### Version
2.3.0-rc2

### Operating system type + version
Windows 10Pro 20H2

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S + MMU2S 3.92 / 1.0.6

### Behavior
MMU2S Single has been set-up as a new physical printer to work the Octoprint. However at the start of the print it does not ask which filament to use. Instead it goes into a filament change routine. At this point it then chooses to load filament 1. The purge line is also short and resembles that of a non-MMU2 printer rather than the longer MMU2 purge line.

* _Steps needed to reproduce the problem_

1) Use Original Prusa i3 MK3S & MK3S+ MMU2S Single as the basis for a new physical printer. Add IP and API Key for OctoPrint Host Type
2) Use this physical printer to print
3) Would expect to be asked which filament number to be used on printer LCD
4) Instead the printer heats up, does bed levelling and then starts a filament change routine
5) Printer chooses filament 1 
6) Printer prints object successfully but displays "F?" on LCD throughout print

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs

[Tape Caddy Outer.zip](https://github.com/prusa3d/PrusaSlicer/files/5750482/Tape.Caddy.Outer.zip)

rtyr, 2020-12-29:
The issue is caused by your custom printer profile, specifically by your start g-code (it looks like it is copy-pasted from some "Pretty" profile). There are no commands to trigger filament selection on LCD and short purge line is also defined in your custom start g-code.

Use stock printer profile or fix your custom printer profiles.

Related commands:
Tx - asks user which filament should be used and loads the filament to the gears.
Tc - loads the filament from the gears to the nozzle (the nozzle should be already preheated before this command).

alienryes, 2020-12-29:
Thanks rtyr. It looks like whereas before I have been able to use the Pretty profiles without overwriting all the physical printer settings like the start g-code, this is no longer the case. Not sure where the behaviour has changed, whether it's me or the slicer but I will follow your advice.

closed 2020-12-29 by rtyr