prusa3d/PrusaSlicer #5624: [Feature Request] Custom failure G-code

inismor, 2020-12-29:
Part of my _End G-code_ sequence presents a print by running the print head up a bit and pulling the bed to the front to easily access it. It also turns off lights, etc.

As often as not sometimes, prints are halted to address issues. When this happens, exactly what I want is the same present behavior: move the printhead out of the way and move the bed forward. But I also want to keep the bed lights on, keep the temperature steady, etc. Today this all has to be done manually.

What I'd like to see is two more custom G-code sequences: **_Success_** and **_Failure_**. I also believe _End G-code_ should run after both of these sequences, regardless. Commands having to do with either sequence of events could be localized to that status. The current _End G-code_ could be migrated to _Success_ and _End G-code_ could become a no-op, but there in case a user might need it.

neophyl, 2020-12-29:
This is a function of a printers firmware not the slicer.  The slicer when its generating gcode only knows about planned actions such as pause or change filament etc.  Even with those it just passes on the appropriate gcode command to the printers firmware.  How the firmware handles those commands is up to it.

When you pause from the printer you are instructing the printer to pause in the performance of following the list of instructions that has already been sent to it by the slicer.  

As such you need to open an issue for the firmware of whatever printer you are using.  Some more modern firmwares like Klipper do allow you to define the actions to be taken upon events like pause and so does running your prints from Octoprint I believe.

bubnikv, 2021-01-04:
firmware request.

closed 2021-01-04 by bubnikv