prusa3d/PrusaSlicer #5627: No support created

Sineos, 2020-12-29:
### Version
Version: 2.3.0-rc2+win64
Build: PrusaSlicer-2.3.0-rc2+win64-202012221641
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042)

### 3D printer brand / version + firmware version (if known)
Modified Ender 5 / Klipper (not relevant for this issue)

### Behavior
* No support is being created on 90° overhang area

[image]
[image]

#### Project File (.3MF) where problem occurs
[no_support.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/5751265/no_support.3mf.txt)
(just remove txt extension)

rtyr, 2020-12-29:
You have "Don't support bridges" enabled. Disable it if you want supports there.

[image]

Closing.

closed 2020-12-29 by rtyr

Sineos, 2020-12-29:
Thanks. Appreciated