prusa3d/PrusaSlicer #5628: Support speed and recognition

MeerkatSuricata, 2020-12-29:
Version 2.3.0-rc2+win64

Win10 x64

For some unknown reason the slicer considers support as a separate object, therefore slowing down it's printing speed as it falls under the rule of "slow down if layer print time is below...". Is there a reasoning behind this behaviour? I want my speed to be consistent throught the layer. Also you can see that the slicer does not notice the support by sliding the layer progression bar (or whatever it's called) to the very end, the support is just greyed out. See the screenshots below for detailed look, also 3mf file is attached as a zip.

[image]
[image]

[image]
[image]
[image]

[Wine_Hand.zip](https://github.com/prusa3d/PrusaSlicer/files/5751691/Wine_Hand.zip)

MeerkatSuricata, 2020-12-29:
ok I have found out that there is a different layer height between the model and the support, but I have no idea why. Ticking off "enable variable layer height" doesn't help, although changing all layer height limits to 0.2 kinda fixes the problem. But what if I want to use variable layer height and keep other layer heights consistent throught the layer (as you can see it's not consistent on the second screenshot)?

[image]
[image]

rtyr, 2020-12-29:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/984. Closing.

closed 2020-12-29 by rtyr

neophyl, 2020-12-29:
Try enabling synchronize with object layers under Print>Support.
PS uses variable layer heights for supports that have no correlation to the model usually as you have discovered.

bubnikv, 2021-02-16:
Fixed with 6ea5781