prusa3d/PrusaSlicer #5629: Corrupted G-Code for Smoothie with PS 2.3.0-rc2

Hardy56, 2020-12-29:
### Version
2.3.0-rc2+linux-x64

### Operating system type + version
Linux 4.15.0-128-generic x86_64
Kubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Modified Bibo2
Smoothieware 12.03.2020

### Behavior
Slicing a quite simple object with version 2.2.0 works like a charm - same object and corresponding STL file with 2.3.0-rc2 produces a gcode file with nearly **double size**. The print is distorted by xy shifts in irregular directions - fail.
Comparing both gcode files shows there are areas with extra lines containing only x and y positions:

[image]

Any GCODE viewer seems to ignore these extra lines - not showing up any error.
But my printer seems to be puzzled by them - don't have a clue why.
Cutting out these suspicious lines and feeding them to Prusa GCODE viewer shows an error (file does not contain valid G-Code):

[image]

May be other printer working with Smoothieware are also affected by this.
As i can go on with previous version it's just blocking me to upgrade to the new version.
Many thanks for your great work on this marvelous piece of software !

rtyr, 2020-12-30:
Please provide your 3MF project file.

Hardy56, 2020-12-30:
Hello,
missed how to provide 3mf ... so i added additional extension:
[Basis_5v3b.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5755041/Basis_5v3b.3mf.zip)

mvadu, 2021-01-01:
I don't have this problem with Smoothie, I am using it with Cetus printer and a tinyfab CPU, which is running [my own build of Smoothie](https://github.com/mvadu/Smoothieware/releases/tag/v2)

Hardy56, 2021-01-03:
@mvadu : thanx for your comment, encouraged me to do further tests.
The additional lines mentioned in the starting post mainly are a result of the activated feature 'avoid crossing perimeters'.
Although this was active too with the PrusaSlicer 2.2.0 tests, it did not produce such extensive positioning drives with my special test model.
Deactivating 'avoid crossing perimeters' in 2.3.0-rc2 results in a successful print !
Because these extra lines contain valid positioning commands only, i still do not understand why Smoothie gets puzzled by that. An update of the firmware to 'Build version: edge-2f4e7e3, Build date: Dec 24 2020 20:06:53' did not show up any difference.

rtyr, 2021-03-01:
Avoid crossing perimeteres feature was completely reworked in 2.3.0, so different g-code output is expected. Seems like smoothieware issue. As you mentioned, those lines contain valid positioning commands. Closing.

closed 2021-03-01 by rtyr