prusa3d/PrusaSlicer #5630: Problem with brim

regnarte, 2020-12-30:
2.3.0 rc2
_Version of PrusaSlicer used goes here_RC2

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Windows 10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

Printed an item with a brim.  It printed fine but the brim was nearly unremoveable.  This does not happen with 2.2.0.  Printed on Prusa MK3S

bubnikv, 2021-01-04:
There was no change in brim betwen 2.2.0 and 2.3.0

rtyr, 2021-03-22:
No response. Closing.

closed 2021-03-22 by rtyr