prusa3d/PrusaSlicer #5633: Adaptive cubic infill weird behavior

abukharov, 2020-12-30:
### Version
2.3.0-rc2+win64

### Operating system type + version
Windows 10 Pro 20H2 19042.685

### 3D printer brand / version + firmware version (if known)
Custom CoreXY, RepRapFirmware for Duet 2 WiFi/Ethernet 3.2-beta4

### Behavior
Description:
I sliced a box with 1cm thick walls using Adaptive Cubic infill. Instead of printing the infill within one layer contiguously PS wants to print a little element on one side of the box then it goes to the other side and then back, which takes a very long time.

Steps to reproduce:
* Model a box with 1cm thick walls
* Load the STL into PS
* Choose Adaptive Cubic infill
* Slice it
* In preview select a layer somewhere in the middle and drag the horizontal slider to see the weird printing sequence (or actually print it and experience this behavior).

Expected result:
Infill is printed contiguously without swinging from side to side.

Actual result:
Printing head goes from one side to the other and back to print tiny infill elements

It's not a new feature request.

#### Project File (.3MF) where problem occurs
[cutlery_tray.zip](https://github.com/prusa3d/PrusaSlicer/files/5753320/cutlery_tray.zip)

https://user-images.githubusercontent.com/21993757/103335797-4962e800-4aca-11eb-88ac-3f10ed706aa1.mp4

Wextor, 2020-12-30:
I have exactly the same issue, this happend with Support Cubic and Adaptive Cubic.

abukharov, 2021-01-02:
Tried playing with the connect settings, didn't make any difference for me.

dattas, 2021-01-03:
I also could not get this issue to disappear with increasing the infill to a higher percent when using Support Cubic. (2.3.0-RC2, 64-bit). There is a ton of unnecessary and wasteful travel back and forth to opposite sides of the model using this infill pattern.

bubnikv, 2021-01-04:
Fixed with 1076e077ce9da0bdc6daa3ab98c135de7cfc00b3

closed 2021-01-04 by bubnikv