prusa3d/PrusaSlicer #5634: [Bug] Ironing produces brittle parts

RichiH, 2020-12-30:
### Version
2.3.0 RC2, Linux

### Operating system type + version
Debian testing/unstable

### 3D printer brand / version + firmware version (if known)
Prusa Mini+, 4.2.1

### Behavior
When printing with ironing enabled, small vertical structures come loose from the horizontal surfaces. This is true for thin walls as well as for columns.

Walls coming loose:
[image]
As an aside, this weird circle-shaped fault only happened with ironing, too. I can't be certain it's caused by ironing, though.

Two times the exact same model, only mirrored, printed with and without ironing, with the columns breaking off with zero/almost no touch:
[image]
Please note that I added filleting to the walls in this file to increase adhesion, but didn't think of doing the same for the columns...

While I can fix this by soldering some filament into place, I would prefer not to need to do this.

#### Project File (.3MF) where problem occurs
[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/5753492/bug.zip) (Without ironing enabled, just add "all top surfaces" to get a broken print)

### Proposed solution / Workaround

Adding filleting to the verticals helps, but is not possible if you only have the STL.

If the problem is ironing under the verticals, not doing that should fix it. If it's a problem with cooling of the plastic too much, adding a warning on models with large surfaces and thin verticals would be an option.

rtyr, 2020-12-30:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/5000 and https://github.com/prusa3d/PrusaSlicer/issues/5113.

stepikovo, 2022-03-23:
As @rtyr mentioned in a different issue, this is most likely caused by a heat creep, for more: https://help.prusa3d.com/en/article/ironing_177488#downsides

closed 2026-09-29 by prusaslicer-bot[bot]