prusa3d/PrusaSlicer #5638: Selecting a Print Settings User Preset that Shares a System Preset Name Selects the System Preset Instead

MDonnellyDev, 2020-12-30:
### Version
2.3.0-rc2+win64

### Operating system type + version
Windows 10 Pro 64-bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S
3.9.1-3518

### Behavior
* _Expected Results_
  * Selecting a print settings user preset with the same name as a system preset from the plater view side tools should enforce the selection
* _Actual Results_
  * When selecting a print settings user preset with the same name as a system preset from the plater view side tools, Prusa Slicer internally loads the system preset instead
  * This can be confirmed by loading the preset and slicing, or by using the plater view to change to the user print settings, and then tabbing to the Print Settings tab
  * Plater dropdown: 
    * [image]
  * Print Settings dropdown
    * [image]

#### Project File (.3MF) where problem occurs
Not really unique to this project, but just in case the AMF helps understand the issue better...
[Uploading phone_case_v2.zip…]()

I would consider this a non-critical bug because two workarounds are available/obvious:

1. Rename user profiles so they don't match system presets
2. Tab to Print Settings to select the appropriate preset, and re-slice (selecting it from this tab does work)

bubnikv, 2021-01-04:
Well, you found another loophole in PrusaSlicer.
We should not allow you to name your user profile that way.

rtyr, 2021-03-01:
Fixed in the master. It will be not possible to save custom profile with the same name. Closing.

closed 2021-03-01 by rtyr