prusa3d/PrusaSlicer #5639: Slicing error (see attached STL)

toastfr35, 2020-12-30:
### Version
Version:   2.3.0-rc2+win64
Build:     PrusaSlicer-2.3.0-rc2+win64-202012221641

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 17,110MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.89
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 3070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* Slicing the attached file generates empty layers.

#### Project File (.3MF) where problem occurs
[image]

[spray_bottle_holder.zip](https://github.com/prusa3d/PrusaSlicer/files/5755007/spray_bottle_holder.zip)

johnny-saigon, 2020-12-30:
I don't think that you can print those holes vertically without supports. The error messages that I get from your model explains it all.

[image]

neophyl, 2020-12-31:
Your stl file is broke.  As you are using thew windows version if you are on windows 10 you can load it into slicer and then right click and choose fix through Netfabb.  This will fix the model for you.  It will then slice fine and will print without support no problem.

As a matter of interest I imported it into Blender and ran its 3dprint tool on it for a report.  It has 144 Non Manifold edges (which are the biggest problem when slicing as all models should be manifold).  It also has 11 Bad contiguous edges, 60 intersecting faces, 434 zero faces, 241 zero edges, 68 non flat faces and 73 thin faces.

toastfr35, 2020-12-31:
Thanks. Looks like the issue is with OpenSCAD then.  Cura did slice it ok in the end, maybe it automatically fixed it.

closed 2020-12-31 by toastfr35

HKalbasi, 2021-02-25:
@neophyl I have same problem. What can I do in Ubuntu?
@toastfr35 Do you report this issue in OpenSCAD?

foreachthing, 2021-02-25:
@HKalbasi… you can use https://service.netfabb.com/login.php or Blender (maybe MeshLab works on Linux) to fix your mesh.