prusa3d/PrusaSlicer #5640: Material used in output filename

taker218, 2020-12-30:
### Version
Version 2.3.0-rc2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Not relevant for this issue

### Behavior
I just started to use PrusaSlicer for slicing my files for a resin printer and wanted to add the used material to the output filename.
I already do this for my FDM printers, so I can see how much filament I need for this print.

Is there a variable for the used material on resin prints as well?

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-01-04:
Will sla_material_settings_id work?

taker218, 2021-01-04:
If I use this placeholder I get the name from my material settings, not the used material (in ml) like in the Sliced Info 
[image]

I may not have been too clear in my post there.

bubnikv, 2021-01-04:
"print_time", "total_cost", "total_weight",
        "objects_used_material", "support_used_material"

taker218, 2021-01-04:
total_weight says zero on my print.

The other two "objects_used_material" and "support_used_material" seem to give me some kind of output, but it's not in ml.

objects_used_material: 1138.77 (should be rounded to 1.14)
support_used_material:1221.86 (should be rounded to 1.22)
total_weight: 0

theslimshaney, 2021-01-05:
It's in mm³. You can convert to ml: `{(objects_used_material+support_used_material)/1000}`
Not sure how/if you can round

taker218, 2021-01-06:
> 
> 
> It's in mm³. You can convert to ml: `{(objects_used_material+support_used_material)/1000}`
> You can also round using math `{(int((((objects_used_material + support_used_material)/1000.0)*100.0-0.5))+1)/100.0}`

That's exactly what I wanted. It looks funky, but it works 👍 
Thanks!

closed 2021-01-06 by taker218