prusa3d/PrusaSlicer #5641: #FeatureRequest - Add post-processing scripts capabilities to SLA printers

X3msnake, 2020-12-30:
FDM printers have this capability
http://projects.ttlexceeded.com/3dprinting_prusaslicer_post-processing.html

Please enable this as well on SLA printer mode in order to use scripts like UV3DP and UVTools to manipulate and convert the sliced file to other formats, or send directly to 3rd party printers

sn4k3, 2020-12-30:
I always wonder why SLA variant drop that box, that feature can open a wide door to manipulate the exported file in any matter. But rather than having a post-processor on "Print Parameters" i think this fits better per "printer", it doesn't hurt to have them in both places although.
Also instead of calling a python script, accept system commands would have a wider usage and not depending on python itself, eg each line executed under a system() command or process. 
The usages are limited to user imagination or needs, but this can be very helpfull, for example to create scripts to upload a file via Wifi to capable printers, a statistic app, convertions, other manipulations.

Area5142, 2020-12-31:
When I used this feature for FDM printing it was not limited to Python, but any executable could be used as post processor for the generated g-code (Linux).

bttguy, 2021-01-14:
+1 from me. Just got a resin printer, and PrusaSlicer looks the best slicer for that also for me. If other formats wont be supported for a while or at all, post processing would be very helpful to convert the format right after exporting. It's a huge miss in PS's popularity that it only support .sl1, but this would help a lot for us, makers! :)

closed 2026-09-29 by prusaslicer-bot[bot]