prusa3d/PrusaSlicer #5648: [FR] Make Elephant foot compensation a filament setting

Sineos, 2020-12-31:
The adaptive Elephant foot compensation is an excellent feature.
In my experience, different filaments are differently prone to producing this effect. PLA much stronger than PETG and ASA produces nearly none on my printer.
I would like to propose to make this a Filament setting rather than a Print setting.

Area5142, 2020-12-31:
In my experience, elephant compensation is more depended on the actual model and z-level than on the filament itself (model has bottom chamfer).
The way you describe the feature request it is more a setting on filament type (PLA, PETG, Flex...) than on a specific filament.

If implemented as a filament override it would work both ways, allowing to make a model level setting and a filament level setting.

locki-cz, 2021-04-25:
I will try make some small prints and show you proof  how it looks like on PLA and PETG with same z-level on same model, any setting on fillament cant fix elefant foot now.

closed 2026-09-29 by prusaslicer-bot[bot]