prusa3d/PrusaSlicer #5650: [Feature Request] Handle small first layer extrusions "better'

vintagepc, 2020-12-31:
### Version
2.3.0-RC2

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
Not printer specific

### Behavior
When prints with complex first layer geometries with small features are sliced (for example, parts with first layer text are notorious for producing a lot of "dot" extrusions on the build plate that are done prior to the text outline), it can produce some very small extrusions that are too tiny to adhere properly to the build plate unless it's absolutely spotless - and even then they may not always stick well and will tend to detach and become lodged in other parts of the first layer. 

[image]

I am thinking it might be a good feature to be able to "de-prioritize" first layer extrusions under a certain total length until after any adjacent longer outer perimeters have been completed. In the above screenshot, for example, the tiny ovoids would be delayed until after the outer perimeter is complete - much like how the gap fill is done as a final stage. (Perhaps "small extrusions" on the first layer could be reclassified to just before a gap fill pass?)

Apologies if there is already a request for something similar - I did try a search but I didn't have any luck coming up with a concise set of "non-generic" terms that found something similar. 

Part in the screenshot is the mantis clamp, sliced with the author's recommended 3 perimeters:

https://www.prusaprinters.org/prints/48505-mantis-clamp

_Is this a new feature request?_ Yes

neophyl, 2020-12-31:
I think I remember a post where someone requested it print outer perimeters first on the first layer and work inwards for exactly this reason.  
Of course I cant find it either now :)

vintagepc, 2020-12-31:
Yes, I think being able to enable that for the first layer only could work as a solution - I didn't want to get too deep into a "how to" approach in my feature request since I'm not familiar enough with the codebase to comment. 

Heh - saw your initial pre-edit you couldn't find it anymore. Good to know it wasn't just me initially having trouble finding the right words for it. :)

Though as noted in that issue that particular option can have other unpleasant side effects relating to nozzle priming and flow - so if practical to implement as proposed (just delaying short extrusions only rather than reversing the order of all perimeters) it would likely result in a better finished part.

vintagepc, 2026-09-02:
Bump because the PS3 alpha still does this on certain geometries and I consider it a valid post-3.0 feature/improvement