prusa3d/PrusaSlicer #5652: RC2+ Program crash when using raft with modified Shape-Box with height 0.5mm

brettvitaz, 2020-12-31:
### Version
Version 2.3.0-rc2+

### Operating system type + version
MacOS Big Sur (11.1)

### 3D printer brand / version + firmware version (if known)
Ender 3 / Prusa MK3S+

### Behavior
Program crashes when using a raft with the Shape-Box

_Steps needed to reproduce the problem_
- Add shape "Shape-Box"
- Resize to 25x25x0.5
- Add 1 layer raft
- Slice

_Expected Results_
- Sliced stubby box with raft

_Actual Results_
- Program crashes

#### Project File (.3MF) where problem occurs
[Shape-Box with Raft.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5758104/Shape-Box.with.Raft.3mf.zip)

lukasmatena, 2021-01-02:
Reproduced, thanks.

For reference: In version 2.2.0, slicing is stopped by the 'Empty layers detected' error, so there is some problem there as well, it probably just stops before the crash. The error was turned into a warning in 2.3.0.

lukasmatena, 2021-01-02:
The problem is that raft is only generated when `((m_config.support_material || m_config.raft_layers > 0) && m_layers.size() > 1)` (PrintObject.cpp:420). In this case there is just one object layer, so the last condition is `false` and the raft layer(s) is then missing. This scenario (one-layer object + raft) apparently never worked (I tried in 2.0.0, 2.1.0 and 2.2.0). It just did not crash.

Given how close the 2.3.0 final is, we will probably just hotfix the crash itself (so we get the behaviour from previous versions) and leave the problem itself to be solved later. I'm not sure why the condition wants two or more object layers, there may be a reason. The hotfix is currently in branch `lm_230_rc2_fixes` (b1a98bd) to be reviewed.

lukasmatena, 2021-01-04:
The crash should be fixed in `master` by 837070a for 2.3.0 final release. The raft is still not generated (same as in previous versions) but it does not crash now. I'll leave the issue open for now.

bubnikv, 2021-02-25:
Fixed with 43c05d35eaf60c52f355db044a38f2147dd695c5

closed 2021-02-25 by bubnikv