prusa3d/PrusaSlicer #5655: Openings in walls when printing in vase mode single perimeters the same width as the "Extrusion width perimeters"

deejay2, 2021-01-01:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
POP!_OS 20.10 (Based on Ubuntu 20.10)
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa i3 MK3S with firmware 3.9.2

### Behavior
* _Describe the problem_
* When printing a test cube in vase mode with 0,42mm thick walls, there are unwanted openings at the bottom.
* _Steps needed to reproduce the problem_
* Create a 4 sided square with 0,42mm width sides and extrude it in Z 10mm in your modeling software.
* Import it into PrusaSlicer 2.3 with 0,42mm "Extrusion width perimeters", enable vase mode, slice it and see for yourself.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* Shouldn't see weird openings?
* 
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

Area5142, 2021-01-01:
This could be an error in the used STL model. Have you tried to create the cube in PrusaSlicer (Right click empty build plate, _Add Shape->Box_?

deejay2, 2021-01-02:
[image]

It fixed the openings but there is still a weird thing happening around the same height as when it was switching from openings to correct.

deejay2, 2021-01-02:
Maybe it's the amount of bottom layers I have set which is 4??

Area5142, 2021-01-02:
In vase mode, the bottom is printed like normal mode and the real vase mode just starts on top of that. What is shown is the start of vase mode and the ramp up.

bubnikv, 2021-01-04:
looks good. closing.

closed 2021-01-04 by bubnikv