prusa3d/PrusaSlicer #5657: Sequential Printing seam alignment not respected

iLikeDarbo, 2021-01-01:
### Version
Version 2.3.0-rc2+linux-x64
PrusaSlicer-2.3.0-rc2+linux-x64-202012221648.AppImage

### Operating system type + version
Operating System: openSUSE Tumbleweed 20201226
KDE Plasma Version: 5.20.4
KDE Frameworks Version: 5.77.0
Qt Version: 5.15.2
Kernel Version: 5.10.1-1-default
OS Type: 64-bit
Processors: 4 × Intel® Core™ i5-3470 CPU @ 3.20GHz
Memory: 7.6 GiB of RAM
Graphics Processor: Mesa DRI Intel® HD Graphics 2500

### 3D printer brand / version + firmware version (if known)
Creality Ender 3, but issue can be observed in PrusaSlicer Preview mode without printing

### Behavior
* _Describe the problem_
  * Sequential printing of multiple copies of the same object. Seam position set to "Aligned". Only the first object has aligned seam positions. For all the subsequent objects, the seam positions gets reset to "Random"
* _Steps needed to reproduce the problem_
  * Place multiple copies of same stl on the Plate. Set Seam Position option to "Aligned" and Slice
* _Expected Results_
  * All objects to have aligned seam positions (aligned_seams.png)
* _Actual Results_
  * Only first object has an aligned seam, all others have random seam placement (random_seams.png)

_Is this a new feature request?_
No

Aligned Seams on first object 
[image]
Random Seams on all other objects
[image]

3MF files and stl file that reproduces issue
[necklace_bead.zip](https://github.com/prusa3d/PrusaSlicer/files/5758968/necklace_bead.zip)

bubnikv, 2022-06-29:
Could you please retest with the upcoming PrusaSlicer 2.5.0-alpha2?

iLikeDarbo, 2022-07-02:
Hi bubnikv. Issue is fixed now, seams are aligned on all objects
Tested on same machine as above using PrusaSlicer-2.5.0-alpha2+linux-x64-GTK3-202206281406.AppImage

thanks for fixing!

closed 2022-07-02 by iLikeDarbo