prusa3d/PrusaSlicer #5658: PrusaSlicer 2.3.0-rc2 crashes often when entering printer information or hitting the Test button

rocketmany2k, 2021-01-01:
### Version
PrusaSlicer 2.3.0-rc2

_Use `About->About PrusaSlicer` for release versions_
2.3.0-rc2+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
i3MK3S + firmware 3.9.2

### Behavior
Every time I want to print directly from the software, I have to enter the printer name, URL for the printer and the key. Often when I hit the Test button or Return after entering the information, the program crashes and quits automatically

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

YuSanka, 2021-01-04:
Thanks for your report. But I can't reproduce your bug. If it's possible, send us a short video. It will be very helpful. 
And explain please what do you mean by
> Please also save all these information so that I don't have to retype them all the time

[image]

When you press the OK button, this information is saved for the "myPrinter" and will be used every time, when you have selected this physical printer

bubnikv, 2021-01-04:
>Every time I want to print directly from the software, I have to enter the printer name, URL for the printer and the key. Often when I hit the Test button or Return after entering the information, the program crashes and quits automatically

As @YuSanka commented, the host information is being saved when you click OK.

How exactly does PrusaSlicer crash? Does it just close? Does it display any error message?

Would you please make a screenshot of the Physical Printer dialog filled in? Maybe you are entering some special characters?

closed 2026-09-29 by prusaslicer-bot[bot]