prusa3d/PrusaSlicer #5659: [Feature request] Flexible Name of the gcode file

wilko22, 2021-01-01:
It would be nice to have in the gcode file name:

Sample:

12pcs_ext4_Name of the print_0.2mm_ASA_MK3SMMU2S_5h15m.gcode

12pcs - how many instances does print include

ext4 - which extruder of MMU2S is used

It would be the best to have option in prusa slicer to chose what is included in output name of the print file.

If I need multiple instances i prepare files with different number of items, depend on how much time I have to print.
Now I have to manually input  this data.

timur-tabi, 2021-01-01:
You can already do this if you enable expert settings.  

Print Settings -> Output Options -> Output filename format.

wilko22, 2021-01-02:
I see where to put, but cant figure out what is the code for active extruders and for the number of instances 
I looked in gcode file and added to output {infill_extruder} but it is always number 1.

Area5142, 2021-01-02:
The place-holders can be found here: [Slic3r placeholders (a copy of the mauk.cc page)](https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page))

sslupsky, 2021-01-12:
There is no placeholder for the number of instances.  This is a long standing request.

clintonefisher, 2021-03-10:
This is a great idea!  
How about a **filaments_used** placeholder that reports in a format familiarized by the Prusa printer UI, ie:  F1, F2, etc..
If, for example, a multicolor print uses F1, F3, & F5, the string would be "F1-F3-F5".
It would be very helpful to know, at a glance, in the filename, which filament(s) will be used.

closed 2026-09-29 by prusaslicer-bot[bot]